richonguzman/LoRa_APRS_iGate #104: Display do not shut down after time-out

iw2ejh, 2024-05-25:
I just installed latest version 2024_05_22 with VSCode (should be the same as the 202_05_23 from web interface?), right procedure FW + Filesystem, my HW is a TTGO LoRa32 V2.1     
the use case is solar powered DIGI + WX     (NO WIFI connection)
I noticed that the display, even if you set to shutdown after 30 second never shut down. (same as before)

richonguzman, 2024-05-27:
test the latest firmware on github as some changes has been made to the partitions and some info may not be saved correctly on your board

Please do all the configurations on Google Chrome

iw2ejh, 2024-05-28:
done, now it work
thanks
-marino

closed 2024-05-28 by richonguzman

iw2ejh, 2024-05-28:
I have to correct my last statement,  the problem is still present
If it connect to wifi, the screen goes blank after the time out
if there is no connection to wifi ( only digipeater) then the screen remain active.
So it seems that the wifi connection is triggering something

iw2ejh, 2024-05-28:
No, I put 30 seconds

 
 

Did you put the timeout to zero secs?

richonguzman, 2024-05-28:
But did it receive a packet in this 30 secs?

iw2ejh, 2024-05-28:
No, without any traffic.

richonguzman, 2024-05-28:
The screen turns on after a packet is received...  it no packet is received it will turns off after "timeOut" time

iw2ejh, 2024-05-28:
Yes, I know, in my case it stay on forever.  When I did the upgrade this morning, I keep it connected to wifi to check it from the LAN,  and the screen was shuting down after the timeout,    as soon as I disabled the wifi and used it as pure digi + WX, then the screen stay on and never shut down.
I noticed there is a similar issue #84:  Display always on · Issue #84 · richonguzman/LoRa_APRS_iGate

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Display always on · Issue #84 · richonguzman/LoRa_APRS_iGate

With the Heltec V3 card the display always remains on, even if the correct option and seconds are selected in th...
 |

 |

 |