richonguzman/LoRa_APRS_iGate #105: TTGO TCP 8001 connection error

leonardinux, 2024-05-26:
Trying to use TTGO with iGate firmware cannot access via tcp/8001 - connection refused 10061 (with APRSIS32, netcat).

Serial output at startup:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Starting Station: IZ5FSA-10   Version: 2024.05.22
Initializing LoRa Module
init : LoRa Module    ...     done!
Not connected to WiFi! Starting Auto AP
Not connected to WiFi! Starting Auto AP
-- Sending Beacon to APRSIS --
-- Sending Beacon to APRSIS --

config:
TNC -> enable tnc server ON

richonguzman, 2024-05-26:
Hi Leo,

did you follow the guide. (in images) of how to setup the program ?

did you also enable the tnc on the board and changed the configuration with the "tnc" switch enabled?

Mane76, 2024-05-26:
>Starting Station: IZ5FSA-10 Version: 2024.05.22 Initializing LoRa Module init : LoRa Module ... done! Not connected to WiFi! Starting Auto AP Not connected to WiFi! Starting Auto AP -- Sending Beacon to APRSIS -- -- Sending Beacon to APRSIS --
> 

Can you confirm that you are connected to a WiFi ? Otherwise TNC connection will not work.

I use the TNC feature to connect via TCP-KISS, works perfect with e.g. udpflex.

closed 2024-06-18 by richonguzman