richonguzman/LoRa_APRS_iGate #106: RADIOLIB_ERR_CRC_MISMATCH / 0dBm / 0.00dB / 0Hz

iw2ejh, 2024-05-27:
CRC errors in Syslog.

I noticed that the data decoded with a CRC error are now stripped of from the syslog.

In the past I could understand and do some analysis on why a crc error was generated, now I just see the message that a crc error occurs but cannot see the original data.

example:
before:  "May  6 19:09:50	192.168.1.72	1 - IW2EJH-11 CA2RXU_LoRa_iGate_1.3 - - - CRC / CRC-ERROR / øIW2MUK-9;>APJTT1,WIDE1-1:>LoRa_APRS_Tracker 2023.12.27éþ3¥ / -140dBm / -19.25dB / 433Hz"

now: "May 27 15:05:09	192.168.1.72	1 - IW2EJH-11 CA2RXU_LoRa_iGate_1.3 - - - Rx / RADIOLIB_ERR_CRC_MISMATCH / 0dBm / 0.00dB / 0Hz"

As you can notice, before I could know the problem was a low signal (-140dBm) and I could correlate it to a Ham ID (even if this id is corrupted) looking at other close reports in the syslog. Now, I just know there is an error but I do not know if it is a disturb, a week signal or a rx error.

Would it be possible to get the same info as before in the syslog? or eventually have a switch to enable/disable the data we get for debug?

Thanks

richonguzman, 2024-05-27:
just checked, latest mod to syslog outputs the CRC error message and SNR RSSI and Freq error as requested, so no mod is needed

iw2ejh, 2024-05-27:
> just checked, latest mod to syslog outputs the CRC error message and SNR RSSI and Freq error as requested, so no mod is needed

Do you say the output to syslog will be back as in the past?  

  I know the crc mismatch is rubbish but in my case it was usefull most of the time, only one or two characters are wrong, so seeing it in the syslog allow me to understand the source of these CRC errors.

richonguzman, 2024-05-27:
CRC erros are that packets could not be validated as 100% correct.... so 1 , 2 or more wrong characters wont add much info as it would SRRI and SNR or FreqError, right?

iw2ejh, 2024-05-27:
For what I think, syslog is a debug tool, so more info we get better it is, we may or may not use a packet with crc errors, if we get it in the log, we may understand something or nothing. If we got nothing in the log, we will never be able to debug anything. If we get SRRI and SNR is already something we can use.
This is my personal opinion, other may think differently

richonguzman, 2024-05-27:
the latest on github has:

CRC Warning, packet (with faulty characters) , SRRI, SNR, Freq Error and all.

please use the latest and test

iw2ejh, 2024-05-28:
Great, just tested and I can see everything
Thanks