richonguzman/LoRa_APRS_iGate #110: WiFi not connecting if iGate is turned on before WiFi AP

1tilen, 2024-06-05:
I have my WiFi router and iGate connected to the same power strip. I noticed that if I unplug the power strip and plug it in iGate would not connect to WiFi. After inspecting everything I noticed that this only happens in case when iGate boots up before my router. 

When I give power to my iGate it turns on its AP which is normal behaviour. My WiFi takes around 1 minute more than iGate to turn on. After the WiFi Timeout which I have set to 10 min iGate does not connect to WiFi. 

In case I plug iGate after my WiFi is turned on iGate normally connects to WiFi and works as it is supposed to.

Could you please check this? 
Thanks!
73 de S54B

richonguzman, 2024-06-05:
there is a solution already in github code (2024.06.04) 

file : /src/LoRa_APRS_iGate.cpp 

line58: uncomment //#define STARTUP_DELAY 5 //min

this will make the igate wait 5 min to try to connect to wifi and wait for the router to have valid internet info

1tilen, 2024-06-08:
I turned on the delay ad it fixed my issue.
Thank you,
Tilen

closed 2024-06-08 by 1tilen