richonguzman/LoRa_APRS_iGate #111: Support for Lilygo 1.6.1

jclifford73, 2024-06-16:
Hi Ricardo.

Is there any chance of this board being supported?    We have quite a few of our early igates using this board and OE5BPA software.  We would like to migrate at least some of them to your code.
Photo attached

73 John GW4BVE

[image]

richonguzman, 2024-06-16:
John,

this board is the same as Lora32v2.1 so flash it with this name and would work just fine.

jclifford73, 2024-06-16:
Thanks for the prompt reply Ricardo.
I tried that with 2 different boards and they lock up on the opening screen.
73 John GW4BVE

richonguzman, 2024-06-16:
> Thanks for the prompt reply Ricardo. I tried that with 2 different boards and they lock up on the opening screen. 73 John GW4BVE

you need to add a lot more info to your claims as I need it to give you a useful answer

meanwhile.. try to flash it with this link with the new webflasher:

https://richonguzman.github.io/lora-igate-web-flasher/installer.html

closed 2024-06-18 by richonguzman