richonguzman/LoRa_APRS_iGate #112: Packet Gated from IS to RF are not sent using proper thirdpartyformat

F4FXL, 2024-06-16:
To avoid looping of packets between RF to IS packets must be sent using the third party format.
See [here](https://aprs-is.net/IGateDetails.aspx)

Currently outgoing packets are sent in a regular frame with just adding TCPIP into the path
Here an example of a wrong frame being sent out, the Gateway is F4FXL-10
`F4FXL-1>APX221,TCPIP,WIDE1-1,F4FXL-10::F4JXQ-9  :Re�u en Lora !{EO}`

Correct formating must be:
`F4FXL-10>APLRG1:}F4FXL-1>APX221,TCPIP,F4FXL-10::F4JXQ-9  :Re�u en Lora !{EO}`

wb2osz, 2024-06-17:
More accurately, there should be * after the IGate name:

`F4FXL-10>APLRG1:}F4FXL-1>APX221,TCPIP,F4FXL-10*::F4JXQ-9  :Recu en Lora !{EO}`

73,
John WB2OSZ

richonguzman, 2024-08-02:
this was implemented a few versions ago

closed 2024-08-02 by richonguzman