richonguzman/LoRa_APRS_iGate #114: USB power monitoring

14FRS711, 2024-06-19:
Hello, and first of all, congratulations on this project which is working wonderfully.
Would it be possible to add a line of code to show that the USB is powered? (Same as internal battery voltage information.)
My module being connected to a solar panel via USB, this would allow me to know its functional status.

richonguzman, 2024-06-19:
Hi, could you be more specific?

its running only through USB and not "internal" battery?

which board is it?

14FRS711, 2024-06-19:
Thanks for the quick feedback.
I use the T3 V1.6.1 model from Lilygo.
My module is powered by 3.7V battery via the JST socket.
My solar panel powers the USB port.
I would like to know if it was possible to send the information in the APRS bubble that the USB port is powered and that it is recharging the battery.

richonguzman, 2024-06-19:
Ah, now I get it, and sadly the board does not give this information:
it does not say which way its powered
it does not measure current of charge 

sorry!

PS: we can build a shunt current measure (not very exact but works) to have this in between solar panel and usb connector, but this is outside of the board "world/unit"

14FRS711, 2024-06-19:
thank you for the answer.
You can actually do this via the external pin by connecting it.

I thought it was possible internally, because I notice the red LED is lit when the USB port is plugged in and the blue one indicates charging in progress. but this may be managed by another chip.

richonguzman, 2024-06-19:
yes, this is managed by the charge controller chip and has no output info to the esp32 "brain" to do something with it

other boards do, like T-beams

14FRS711, 2024-06-19:
great, thanks for the quick responses.
So I'm going to look for a place to collect the voltage coming from the USB and bring it back to port 34.
thanks again, I would like to learn more about programming like you, nice work)

S57PNX, 2024-06-19:
One of my gates (S58W-13) is similarly powered by a 12V solar system. I have a small 12V to USB converter that feeds the USB port of lora32 2.1.6, and a backup battery via the onboard JST connector. I take the 12V voltage _before_ the DC-DC converter and route it via a voltage divider to pin34. This way, I can see the voltage of my solar system as External voltage and the voltage of the battery as Battery voltage.  If the solar system dies (bad weather), the beacon shows External=0V and Batt=whatever (slowly dropping over time). Hope this helps you design your solution.

iw2ejh, 2024-06-19:
I too have a solar digi, and I have a 5 volts solar panel, that is feeding directly via a charge regulator the internal battery (I do not use the usb port) I can mesure the internal battery voltage and see how it goes,  and I plan to mesure the output of the solar panel directly using the Pin34 (external voltage) In this way you can see the battery voltage and the solar panel voltage.

14FRS711, 2024-06-20:
Thank you for your tips.
In my case I want to limit the number of consumers and the cost.
I tried today to bring the Vcc Usb (from terminal 4 of the 65B311 charging chip) to terminal 34.
without the supply I have my battery voltage at 4.1V and external voltage at 0.55V.
But when the USB is powered I have 6.89V on the battery and 16.08v externally.

richonguzman, 2024-06-20:
YOU NEED TO CONTROL THE VOLTAGE!

esp32 pins only work till 3.3V (if you get higher you risk to burn the pin or the board!

richonguzman, 2024-06-20:
but it cant be higher than 3.3V!!!!

5V could burn the pin (and also the board)

change the voltage divider resistor to avoid this!!!!

14FRS711, 2024-06-20:
[image]

S57PNX, 2024-06-20:
> The voltage is precisely 5V at the input of terminal 34. This is why I don't understand the display problem.
> Perhaps a problem with looping back the GND which distorts the result?

Pin34 only handles up to 3.3V, you are killing it with 5V.You need a voltage divider to bring your 5V under 3.3V. The code assumes a 100k/27k divider.Sent from my iPhone

14FRS711, 2024-06-20:
I thought 5v was the max :(
So I'm going to put a divider.
Fortunately the card is strong
Thank you for this warning.

richonguzman, 2024-06-20:
> I thought 5v was the max :( So I'm going to put a divider. Fortunately the card is strong Thank you for this warning.

Arduino pins can hold 5V , esp32 only 3.3

14FRS711, 2024-06-28:
Thanks for the information, I now understand. I therefore installed a resistor of 100k and the other of 58k for my input voltage at 5.2V max.
Thanks to the update this avoids having to modify the source code for the R values.
Thank you

richonguzman, 2024-06-28:
V1.0.4.2 is available with Resistors values input from WebConfig

closed 2024-06-28 by richonguzman

14FRS711, 2024-06-28:
yes exactly thank you

14FRS711, 2024-06-28:
[image]