richonguzman/LoRa_APRS_iGate #115: BME280 not work

kecsa, 2024-06-20:
Hi everybody,
I use heltec_wifi_lora_32_v3 with firmware 2024_04_28. works very well.
I uploaded 2024_06_18, it also works fine, except for the BME280 sensor. In the .json file, BME is "active : true", BME is connected to SDA=17, SCL=18 on the board, in parallel with Oled, but alsos nothing related to BME280 in the VSCode monitor window during booting.
There are 2 new lines in boards_pinout.h: BOARD_I2C_SDA and BOARD_I2C_SCL. If these were in their original state,0 41 and 42, the boot is fast, but there is no BME280, if I rewrite them to 17 and 18, the boot stops approx. for 40 seconds and still no BME280.
Where could the problem be?
Many thanks.

richonguzman, 2024-06-20:
Hi,

Heltec V3 had problems most of the times when running I2C modules in parallel with OLED pins

this is why it is implemented (for a while) with a secondary I2C pins

41 SDA
42 SCL should be the connections for your BME280

check the connections please and let me know if this worked for you

kecsa, 2024-06-20:
Thanks for the very quick response.
I just soldered it to 41 and 42 and the BME280 works perfectly..
(I didn't think about that because in the original state, Oled + BME280 .... 17,18 with version 2024_04_08, the secondary I2C pins were not needed)
Thank You very much , everything is working fine.

richonguzman, 2024-06-20:
great!

closed 2024-06-20 by richonguzman