richonguzman/LoRa_APRS_iGate #116: internal and external voltage as Telemetry

iw2ejh, 2024-06-22:
Would it be possible to format the value from the internal voltage and external voltage in a way it could be sent as Telemetry so that it could be displayed in the Telemetry page of the various APRS-IS clients, so we may have a trace on the value during days & weeks?
Thanks for all the effort you are putting in the development of this project.
73 - IW2EJH

richonguzman, 2024-06-22:
Sending as telemetry won't be that much complicated when sended over TCPIP/Wifi/internet 

But sending telemetry is sending two packets 

And this may generate more rf usage as is now

Or is this expected?

iw2ejh, 2024-06-22:
Yes sure it is expeted, Usually the parameter packet is sent once every 24 or 48 hours or more, the data packet can be sent based on need, 30 minutes, 1 hours, or other, depending if it goes via TCP or via RF, this occurs also in vhf APRS mode. Telemetry is indipendent of the beacon and ad some traffic, should be used if need.

richonguzman, 2024-06-25:
so how do you propose this to be implemented ?

send a gps-location (with or without Wx data) at 15 min and besides this another packet after the gps-location with the current battery info over telemetry format?

iw2ejh, 2024-06-25:
I would say that the telemetry has nothing to do with the wx data. This telemetry report various info depending of what we want to control.
The Parameter, Name and Equation Coefficients message may be sent once, or on reboot, as it do not change and is used to understand the data received, the value of the telemetry could be sent setting a timer and depending if it goes only via internet, or if it will be sent via RF.
Here a link reporting some evolution of the original  APRS Telemetry implementation:  https://blog.aprs.fi/2020/03/aprsfi-supports-kenneths-proposed.html

richonguzman, 2024-06-25:
so, I created this https://github.com/richonguzman/LoRa_APRS_iGate/discussions/122

to keep with the discussion of values for telemetry