richonguzman/LoRa_APRS_iGate #117: OLED display AlwaysOn

JayP0100, 2024-06-23:
Unable to turn off OLED display by inhibit "Always On"
I suspect something else may be keeping it alive
The intended use is a solar powered i-gate that will only digipeat to other locations with WiFi
No Wifi has been configured so always in "Auto AP" mode; is this keeping the display on?

I confirmed that "alwaysON" is "false" in the backup of the json file with "Display Timeout" of "60"

Physical board is Lillygo Model T3 V1.6.1 and I selected ttgo-lora32-v21 when applying the initial flash

richonguzman, 2024-06-23:
Hi

Put time to "0" and try

JayP0100, 2024-06-23:
I'm new to github and after I left the issue I started reading the closed issues (I did go thru the 17 open first!)  Seeing the similar issues in the closed category, does entering "0" prevent the OLED from coming on at all?  

I was hoping to see the OLED to confirm operation on initial power up then have the display extinguish for lower power consumption when in digipeater mode and solar powered

Thanks

JayP0100, 2024-06-23:
Zero does indeed extinguish the OLED right after briefly displaying Starting Auto AP.  I think that will suffice for now.  Appreciate the quick response and all your work on this project

closed 2024-06-23 by JayP0100