richonguzman/LoRa_APRS_iGate #119: Heltec WSL V3 with BME280 sensor is sending "g...t...r...p...P...h.. b...."

pamidev, 2024-06-23:
I'm trying to connect a BME280 sensor to Heltec WSL V3 using pins 3.3V or 5V, GND, no. 41 and no. 42 - instead of telemetry there is "g...t...r...p...P...h.. b....". When I connect the same sensor to the pins on Raspberry Pi 4B, the sensor works fine. Can anyone test the BME280 sensor with this board and can confirm its operation? I don't know if it's a problem with my board or with the firmware. How to check?

richonguzman, 2024-06-24:
Hi,

if you start the board and have any serial output app running you will see some info in the boot up process

this will show if the boards is not recognized or its a code thing

can you test inverting SDA/SCL pins ?

S57PNX, 2024-06-24:
Did you use the Blue logo with L in the settings? Aprs.fi will not detect telemetry and display this strange text in the comment instead. Only the stations with Blue logo will be parsed by aprs.fi as weather stations and parsed accordingly.

richonguzman, 2024-06-24:
> Did you use the Blue logo with L in the settings? Aprs.fi will not detect telemetry and display this strange text in the comment instead. Only the stations with Blue logo will be parsed by aprs.fi as weather stations and parsed accordingly.

if Wx data Tx is enable, it will/should change the logo into Blue + L automatic.

pamidev, 2024-06-26:
> Hi,
> 
> if you start the board and have any serial output app running you will see some info in the boot up process
> 
> this will show if the boards is not recognized or its a code thing
> 
> can you test inverting SDA/SCL pins ?

I'll check it and give the answer.

pamidev, 2024-06-29:
I checked. I tried swapping pins 41/42 as SDA/SCL. I also changed the BME280 power supply from 3.3V to 5V. Unfortunately it does not work. Here are my logs:

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

Starting Station: SP8MDH-4   Version: 2024.06.28
Initializing LoRa Module
init : LoRa Module    ...     done!

Connecting to WiFi 'LoRaP' ...
..Connected as 192.168.2.203
[  3863][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3871][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3877][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3883][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3889][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3896][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3902][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3908][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3914][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3920][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3926][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3932][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3939][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3945][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3951][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3957][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3963][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3969][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3975][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3982][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  3988][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  3994][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4000][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4006][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4012][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4018][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4024][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4031][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4037][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4043][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4049][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4055][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4061][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4067][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4073][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4080][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4086][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4092][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4098][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4104][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4110][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4117][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4123][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4129][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4135][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4141][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4147][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4153][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4160][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4166][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  4172][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  4178][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[... 179 lines cut ...]
[  5283][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5289][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5295][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5301][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5307][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5314][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5320][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5326][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5332][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5338][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5344][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5350][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5357][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5363][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5369][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5375][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5381][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5387][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5393][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[  5399][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[  5406][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
Connecting to APRS-IS ...     Connected!
(Server: euro.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
---> LoRa Packet Tx    : SP8MDH-4>APLRG1,WIDE2-2:!L4LlDT4$K_ xG.../...g...t...r...p...P...h..b.....LoRa APRS Digi/iGate 434.855MHz @1k2`

S57PNX, 2024-06-29:
SP8MDH-4 Is a green icon on aprs.fi. Switch it to blue L and try again - green will not display weather info. 

(Maybe this is not the only problem that you have, but making it a blue icon manually will take away one uncertainty and allow you to focus on real problem. While the firmware should automatically change to blue, do it manually for now.)

Sent from my iPhone

> On 29 Jun 2024, at 17:03, Paweł Miduch ***@***.***> wrote:
> 
> SP8MDH-4

richonguzman, 2024-06-29:
5V BME may be not a good idea as pins on ESP32 need 3.3V reference only

on the other side, does it show any Wx data on screen? or it does not read BME module at all?

pamidev, 2024-06-30:
> SP8MDH-4 Is a green icon on aprs.fi. Switch it to blue L and try again - green will not display weather info.

Now it is L green, but when testing it was blue, because I switched back. I still have a second LoRa SP8MDH-3 transmitter and there the MeschCom system is sending telemetry data correctly, despite the green icon. See https://aprs.to/

I have a Heltec Wireless Stick Light V3, which does not have a monitor, so I cannot check. The only comment it sends to APRS-IS is:

" g...t...r...p...P...h...b.....LoRa APRS Digi/iGate 434.855MHz @1k2".

iw2ejh, 2024-06-30:
g...t...r...p...P...h...b..... ==> the fields are empty, so no valid value are sent. May be the BME is not sending data to the esp32 at all are you sure the module is working fine and communication is going on? did you try to power the module with 3.3volts?

richonguzman, 2024-07-01:
> > SP8MDH-4 Is a green icon on aprs.fi. Switch it to blue L and try again - green will not display weather info.
> 
> Now it is L green, but when testing it was blue, because I switched back. I have a second LoRa Digi/iGate in the MeschCom system as SP8MDH-3 and there is sending telemetry data correctly, despite the green star. See https://aprs.to/station/5247635
> 
> > on the other side, does it show any Wx data on screen? or it does not read BME module at all?
> 
> My BME280 module can be powered by 3.3V or 5V because it has a voltage regulator. I have a Heltec Wireless Stick Light V3, which does not have a screen, so I cannot check. The only comment it sends to APRS-IS is:
> 
> `g...t...r...p...P...h...b.....LoRa APRS Digi/iGate 434.855MHz @1k2`
> 
> Maybe it's a problem with powering the BME280 module from Heltec WSL V3. I will try to power it externally (without using Heltec WSL V3) and test it again.

Telemetry and Wx packets are not the same.

on the powering issue, power it with 3.3v from ESP32S3 would be advisable.

steffenspies, 2024-08-01:
I also use the Heltec WSL V3 and BME280 and see the same behaviour - exactly the same output, while the same module with a T-Beam works fine. 

Something I haven’t checked yet, because I‘m not home, is bus termination. The T-beam has a display on the same bus, the WSL is BME only. 

I thoughts?

Steffen, DL1SX

richonguzman, 2024-08-02:
lets start again:

how are you connecting the I2C board? which pins?

steffenspies, 2024-08-02:
Board: Heltec Wireless Stick Light V3
BME280 connected to Pin 41(SDA) and Pin 42(SCL), powered with 3V3. 

BME280 Board is this one
[image]

When switching "Activate Wx Telemetry" on, the serial monitor shows the exact same output as shown in the message above.

richonguzman, 2024-08-02:
you are doing exactly as expected and should have good data with it. Is the board BME good?

steffenspies, 2024-08-02:
It is.
I tried two and put the first one on a T-Beam where it works great. 

The only thing I can think of is that the I2C termination is missing/wrong. 

On the T-Beam is a display also connected to the same I2C bus while on the WSL V3 only the BME280 is connected to.

richonguzman, 2024-08-02:
found the problem! had a missing wire.begin for this board!

please check it (just uploaded)

steffenspies, 2024-08-02:
I had to add adafruit Si7021 Library now, but get this error later on

richonguzman, 2024-08-02:
I added a process to find the board.... I cant test it locally as I have not the board here free

but it should find the bme board now without any issues

steffenspies, 2024-08-03:
I've made a fresh pull and this time it compiled fine.
I get this on the serial terminal:
`BME/BMP/Si7021 sensor Active in config but not found! Check Wiring`
This is now looking promissing and the error messages from above are gone.
Will check this when being back home and will come back.

First of all thank you for the fix, the project and your support in general.

73, Steffen DL1SX

steffenspies, 2024-08-10:
Ok, back at home, I did connect a Logic Analyzer to the I2C bus.
What I see: you try to find the sensor by scanning the addresses from 0x01 up to 0x76 and that's exactly what's happening.
The "ack" is always false, accept for 0x76 where it is true (which is the BME280 sensor).

I've zero to none experience with I2C so maybe these screenshots might help.
[image]
[image]

richonguzman, 2024-08-10:
so the board is found on the pins you connected...  is it still showing any problem?

steffenspies, 2024-08-10:
Yes, I still get this:
BME/BMP/Si7021 sensor Active in config but not found! Check Wiring

steffenspies, 2024-08-11:
I did the same on a T-Beam instead of the Wireless Stick Lite where the sensor is found and working.
`BME280 sensor found
BME280 Module init done!`
As you can see it tries the different addresses to find the sensor until 0x76. That's the same as with the WSL. But on the WSL it doesn't start to read the data from the the sensor.

![Screenshot 2024-08-11 054901](https://github.com/us
[image]
er-attachments/assets/1432bd12-08f9-42e4-bd1d-ee4c68e764ab)

steffenspies, 2024-08-12:
I don't have deeper coding experiences, but did a change in bme_utils.cpp and added the two HELTEC_WSL_V3 devices.
Now it is working fine!

`BME280 sensor found
BME280 Module init done!`

`    void setup() {
        if (Config.bme.active) {
            getWxModuleAddres();
            if (wxModuleAddress != 0x00) {
                bool wxModuleFound = false;
                if (wxModuleAddress == 0x76 || wxModuleAddress == 0x77) {
                    **#if defined(HELTEC_V3) || defined(HELTEC_WSL_V3) || defined(HELTEC_WSL_V3_DISPLAY)**        <<<--------- added WSL
                    if (bme280.begin(wxModuleAddress, &Wire1)) {
                        Serial.println("BME280 sensor found");
                        wxModuleType = 1;
                        wxModuleFound = true;
                    }
                    #else
                        if (bme280.begin(wxModuleAddress)) {
                            Serial.println("BME280 sensor found");
                            wxModuleType = 1;
                            wxModuleFound = true;
                        }
                        if (!wxModuleFound) {
                            if (bme680.begin(wxModuleAddress)) {
                                Serial.println("BME680 sensor found");
                                wxModuleType = 3;
                                wxModuleFound = true;
                            }
                        }
                    #endif
                    if (!wxModuleFound) {
                        if (bmp280.begin(wxModuleAddress)) {
                            Serial.println("BMP280 sensor found");
                            wxModuleType = 2;
                            wxModuleFound = true;
                        }
                    }
                } else if (wxModuleAddress == 0x40) {
                    if(sensor.begin()) {
                        Serial.println("Si7021 sensor found");
                        wxModuleType = 4;
                        wxModuleFound = true;
                    }
                }                
`

richonguzman, 2024-08-12:
> void setup() { if (Config.bme.active) { getWxModuleAddres(); if (wxModuleAddress != 0x00) { bool wxModuleFound = false; if (wxModuleAddress == 0x76 || wxModuleAddress == 0x77) { **#if [...]

@steffenspies can you post a picture of the code you pasted?

it does not clearly say what you have moded to make it work

I'm right to think you pushed it to work as Wide1 ?

steffenspies, 2024-08-13:
@richonguzman Yes, in bme_utils.cpp you have already made an ifdef for HELTEC_V3  to use Wire1. 
I only added the HELTEC_WSL_V3 and the HELTEC_WSL_V3_DISPLAY to it in Line 61

[image]

richonguzman, 2024-08-13:
great, its added and will be in a few hours in new Release ;)

closed 2024-08-13 by richonguzman