richonguzman/LoRa_APRS_iGate #121: Syslog messages don't show up in Visual Syslog Server nor WinSyslog

davidebeatrici, 2024-06-24:
Using WireShark I noticed unprintable characters and a wrong timestamp:

[image]

Possibly related: lora-aprs/LoRa_APRS_iGate#227

iw2ejh, 2024-06-25:
what you see after CRC-ERROR are the data not correctly decoded by the LoRa Radio, it include any kind of characters as it is not parsed. These data are useful for debuging. If you notice, the station you are receiving is IK2UUJ and the level is -142dBm, very marginal. This CRC-ERROR is due to the location and distance, in other cases if you see the level higher it may be due to disturb or colision.

davidebeatrici, 2024-06-26:
Alright, that's what I thought.

According to your screenshot you're receiving the messages just fine in Visual Syslog Server. Is there anything that you had to change in the configuration? Perhaps the filters?

richonguzman, 2024-06-26:
> Alright, that's what I thought.
> 
> According to your screenshot you're receiving the messages just fine in Visual Syslog Server. Is there anything that you had to change in the configuration? Perhaps the filters?

I guess the problem with timestamp is on the syslog server more than on the igate firmware as it just uploads the log but the log does not push time to it. Its the syslog server that saves his own timestamp into the log.

richonguzman, 2024-06-26:
as far as I've tested and others told me, the timestamp is added on the syslog server side.

not on yours?

davidebeatrici, 2024-06-26:
Right now I don't see any messages from iGate, but I'm pretty sure the timestamp is indeed added by the Syslog server.

However, appending the timestamp from the firmware's side would be useful as a reference, especially when there's high latency involved.

richonguzman, 2024-06-26:
> Right now I don't see any messages from iGate, but I'm pretty sure the timestamp is indeed added by the Syslog server.
> 
> However, appending the timestamp from the firmware's side would be useful as a reference, especially when there's high latency involved.

how much latency are you experimenting?

davidebeatrici, 2024-06-26:
Nothing too extreme, but in certain environments it may be useful to know exactly when an event happens.

It's just a suggestion, not something I actually need.

richonguzman, 2024-06-26:
will save this idea for later (when I add NTP server time to igate)

meanwhile I'll close it.

closed 2024-06-26 by richonguzman

davidebeatrici, 2024-06-26:
Sounds good!