richonguzman/LoRa_APRS_iGate #123: Crash after SAVE with option Send External Voltage/Monitor External Voltage Health

DOKTORKROSS, 2024-06-25:
If I turn on one of these options Send external voltage or Monitor external voltage status with only the microUSB power supply connected, the board becomes dead. The OLED shows the last status, no button response. After unplugging and plugging in the power supply, the OLED no longer shows anything, the board is dead.

LILYGO®TTGO LoRa32 V2.1 _ 1.6 wersja 433Mhz ESP32
https://pl.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.35.3a541c24j4pvxv&gatewayAdapt=glo2pol

Station:
https://aprs.fi/info/?call=HF5F-2

73!

richonguzman, 2024-06-26:
With only sending internal battery voltage (and no other switch from Battery Menu activated) it still does not boot?

DOKTORKROSS, 2024-06-27:
I check one by one - problem after enable **Monitor External Voltage Health**. Device not booting.

[image]

richonguzman, 2024-06-27:
did you read the [https://github.com/richonguzman/LoRa_APRS_iGate/wiki/02.-WebUI-iGate-Configuration#battery](url)

where it says that when activating external battery readings , if the value is bellow the value in WebUI it will sleep?

DOKTORKROSS, 2024-06-27:
Could you help me and show how I can connect divider? I tried to find in the images catalog but found nothing.
I do electronics but with these boards i have no experience and i would not like to make a mistake. Many thanks.

"_Send External Voltage from external batteries (Max 15V with predefined 100K+27K Voltage divider)._"

After you suggestion I try check this option without external power but can't set 0V (minimal value is 3V).

richonguzman, 2024-06-27:
[clickHere](https://www.allaboutcircuits.com/uploads/thumbnails/voltage-divider-main-circuit.png)

Vin = 12V battery
R1 = 100K
R2 = 27K
Vout = the ESP32 pin you select to measure the voltage

DOKTORKROSS, 2024-07-02:
I tested. after connecting an external voltage source and saving the Monitor **External Voltage Health** option the board behaves normally. 
[image]

After disconnecting the external voltage source the board goes to sleep after some time. It can still send information about the lack of voltage.
[image]

**Note** that with the external voltage disconnected (only the voltage divider itself) the result is still not 0.00 - perhaps a control for "External Voltage divider Offset" is needed - #129?

After reconnecting the external voltage the board does not start by itself, unless you turn it on using the button (short press). If this was the intention, it is OK. I expected that in standby mode maybe it monitors something and after restoring the voltage it will start by itself ;)

The board during testing I had a micro USB power supply - I did not have a battery but it will come soon.

My parameters.
[image]

Only I would like to suggest adding a delay of 1min for the activation of the function after restart. This will help unwary users to save themselves without reloading firmware or connecting external voltage, which they may not have immediately prepared.

Other than that, the application is closed. Thanks

richonguzman, 2024-07-02:
can you explain the last part of the delay you proposed?

as for the Sleep after low external voltage, the board sleeps 30 min and checks again if it was recharged so it should wake up automatic and reconnect (test it ;) )

DOKTORKROSS, 2024-07-02:
1. If I am a "stupid user" and without external power plugged in I turn on its monitoring and **SAVE** it will make it impossible to get to the option to return - **uncheck this option** back easily (http server does not work, because the board is dormant). I can't restart the board with a button. If I unplug and connect the main power supply I think the board at startup checks if the power supply is connected (?) and immediately goes to sleep so I have no way to change that. The only option is to upload a clean firmware and upload the configuration (as I did) or prepare a battery with a divider and plug it in.

So I suggest that the **External Voltage Monitoring** should be done 1 minute after connecting the power supply, so that the http server starts - which will give me time to quickly **disable** these options and **SAVE** again.

I hope its clear now - if not - ask me again.

2. I will test - i need more than 30 min :)

richonguzman, 2024-07-02:
warning added to wiki

https://github.com/richonguzman/LoRa_APRS_iGate/wiki/02.-WebUI-iGate-Configuration#battery

closed 2024-07-02 by richonguzman