richonguzman/LoRa_APRS_iGate #124: Feature Request - Power fail recovery

jclifford73, 2024-06-26:
Hi Ricardo

Situation
Remote or inaccessible igate.  Power failure and subsequent restoration

Issue
If the wifi AP takes longer than the igate to recover on power restoration, the igate does not connect to the recovered AP.

Possible solutions
1. Parameter in the config file to specify a delay on boot before attemting to connect to wifi
2. If the wifi does not connect and the autoAP is not used, then go back and retry the wifi connection.

env: heltec_wifi_lora_32_V3

Many thanks for your incredible work on this software.

73 John GW4BVE

richonguzman, 2024-06-26:
Hi, on the firmware there is a delay:

LoRa_APRS_iGate.cpp line 58

but this is enabled over VSCODE firmware upload only (for now) please test

jclifford73, 2024-06-26:
OK many thanks Ricardo.  I will edit and test.   I do all the uploads via Vscode.

73 John GW4BVE

jclifford73, 2024-06-30:
Recovery after power failure is now perfect.  The switch and AP come up then the igate connects.   I only need a 1 min delay.   

I have closed the issue but would it be possible to move this configuration to the igate_config.json file?

Many thanks Ricardo

73 John GW4BVE

closed 2024-06-30 by jclifford73

richonguzman, 2024-06-30:
i will try to add this to next release ;)