richonguzman/LoRa_APRS_iGate #127: web page : Received packets

iw2ejh, 2024-06-27:
How do we see the other 10 last received packet in the webpage?   I can only see  the older 10 and not the newest 10.  and in order to see the last received packet I need to wait that it became the 11th.  I tested with firefox, chromium, chrome.  
[image]

ArturStarz, 2024-07-17:
I confirm, I have the same

ArturStarz, 2024-07-18:
I checked - I always have 12 and the 8 newest ones are missing.
[image]
my friend has the same 12/8.

richonguzman, 2024-07-18:
> I tested today.. I always have 12 and the 8 newest ones are missing. <img alt="image" width="1005" src="https://private-user-images.githubusercontent.com/37155239/350090842-6a489fcc-a5e4-4a99-aabb-11afa1fee44e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzMjk2MTQsIm5iZiI6MTcyMTMyOTMxNCwicGF0aCI6Ii8zNzE1NTIzOS8zNTAwOTA4NDItNmE0ODlmY2MtYTVlNC00YTk5LWFhYmItMTFhZmExZmVlNDRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDE5MDE1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxY2EyMGUzNjc4ZDZlYmJlNTlkYzg3N2Q1MzJkYTRlMWJkMTk5N2Y2YjQ5MjQxMGUxZmM3NTE2NDg2ZjhhNzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WVR1Oh7c9Mh2wTEKrEaaAbBg7U5ryUQ5HmfIP-Wa5GU"> my friend has the same 12/8.

thanks for the testing, will check how to solve this

ea5wa, 2024-08-03:
Sometimes doesn't show anything and it's necessary to reboot

Gracias Ricardo por todo el trabajo

richonguzman, 2024-10-21:
I changed the buffer size of it and also added NTP time for real packet rx time. It should be solved

closed 2024-10-21 by richonguzman