richonguzman/LoRa_APRS_iGate #129: UX: BATTERY section layout - group Internal and external voltage options

DOKTORKROSS, 2024-06-28:
I'd like to ask you about group this options as my example if no problem of course.

**Current layout:**
[image]

**My two proposal layouts:**
[image]

richonguzman, 2024-06-28:
Check how it's now and let me know

DOKTORKROSS, 2024-06-28:
I see the last build from 2024.06.25. Stupid quwstion because I didn't understand if I should wait for the next release or compile on my own and confirm if that's what I meant?

[image]

Regards
---
HF5F | Pawel |  Poland

richonguzman, 2024-06-28:
this was meant for you to compile and upload from VSCODE with github new info as I have to build the process to update the web installer still

and I always give a few days of tries before this to validate all with more users if available

DOKTORKROSS, 2024-06-28:
Sure. I'll be compiling and testing over the weekend (same as issue #123).

DOKTORKROSS, 2024-06-30:
It looks very cool. The divider configuration I was just about to report to you, so you read my mind well.
[image]

Q1: I see that some of my colleagues who want to use your software have no technical knowledge. They won't know where R1 and R2 are - I suggest adding a schematic. Ideally, you should put a pin configuration box next to the place on the schematic where it is drawn.

Q2: Not all pins on the board are analog and cannot measure voltage. Maybe find a common part of the pins to select from the supported boards and only allow those to be selected from the list?

Q3: Is pin 35 reserved? It is analog and unfortunately after configuring it did not show anything.
[image]

Q4: Could Monitor External Voltage Health start 1 minute after startup? This is probably my case from #123 - after accidentally turning it on, the only option is to reload the firmware or connect the voltage source to the pin that was configured - you have to remember it :)

Meybe as this?
[image]