richonguzman/LoRa_APRS_iGate #134: After the update mine no longer works.

hem0773, 2024-07-06:
Hello !
after update my call HEMX73 no longer works.

I don't have a license and only want to use the I-Gate to listen

DOKTORKROSS, 2024-07-07:
Where did you get the HEMX73 sign from?
You wrote "_...after update my call..._" but what was the value before?

hem0773, 2024-07-10:
I came up with HEMX73 or what else are people without a license supposed to do?

as it says, only reception, no transmission 

In an older version of the software this worked without any problems

richonguzman, 2024-07-10:
> I came up with HEMX73 or what else are people without a license supposed to do?
> 
> as it says, only reception, no transmission
> 
> In an older version of the software this worked without any problems

Test now the 2024.07.10 version which would enable Non Ham to only Rx with the "callsign" you have

hem0773, 2024-07-12:
Version 2024.07.10 which one is that please? I tested the latest one and then it gets stuck when booting with an error that the call is invalid

richonguzman, 2024-07-12:
> Version 2024.07.10 which one is that please? I tested the latest one and then it gets stuck when booting with an error that the call is invalid

the last one on github repository (not on web installer)

closed 2024-08-02 by richonguzman

Geksogengo, 2024-08-04:
My current callsign R1ZAAG is not validating, can this be fixed?