richonguzman/LoRa_APRS_iGate #135: Question: restore from config file

DOKTORKROSS, 2024-07-07:
Is the configuration file that I can generate from the **Backup/Download** menu only ever valid for one version and not backwards compatible?
I tried to quickly restore the configuration from the **Backup/Restore** menu but unfortunately it didn't work.

I uploaded the latest firmware from the web installer 1.0.4.2, and before that I had version 1.0.4.

richonguzman, 2024-07-08:
im looking at ways to make it fully backwards compatible but the mods get always in the way

DOKTORKROSS, 2024-07-08:
It seems to me that the file structure has changed.
Maybe if the new features was always at the end of the file no matter where it is in the WEB UI it would help?

richonguzman, 2024-07-08:
> It seems to me that the file structure has changed. Maybe if the new features was always at the end of the file no matter where it is in the WEB UI it would help?

this was a "once" mod needed

and new things will get in the bottom as expected ;)

closed 2024-08-02 by richonguzman