richonguzman/LoRa_APRS_iGate #137: Switch between battery and USB

PorterRyan, 2024-07-14:
Perhaps the answer to this has already been made, but I've got a USB solar panel (5W panel with USB output) plugged into my Lilygo Lora32 igate, with a 3.7v battery also connected for backup power. The battery is fully charged, but when I unplug the mains charger and connect the solar panel, the board shuts off. 

I think it is trying to power the board from the low voltage off the solar panel rather than switching over to using the power from the battery while the sun is down.

Is this something that can be fixed in firmware or do I need to make some modifications? I'm aware of https://github.com/richonguzman/LoRa_APRS_iGate/issues/114 and have some resistors on order, but I was under the impression battery vs USB power switching was a separate thing.

S57PNX, 2024-07-15:
The transition from USB to onboard battery and back should be seamless out-of-the-box, it works for me. Did you check the little power switch on the side of the board? It doesn't impact the usb power, but it switches the battery on and off.

PorterRyan, 2024-07-16:
Thanks, it was that simple! I had completely forgotten that the switch even existed.

closed 2024-07-16 by PorterRyan