richonguzman/LoRa_APRS_iGate #139: Add support for YAAC

sysadminpower2019, 2024-07-19:
Hello,

here is another multiplatform client that would be great to get to work with the LORA module.
I tried to set it up like the other examples here but i can't get it to connect neither bia TCP/IP nor via Serial. 

https://www.ka2ddo.org/ka2ddo/YAACdocs/config_kissovertcp.html

richonguzman, 2024-07-19:
there is Serial output and also KISS output (TNC2 encapsulated)

did both not work?

sysadminpower2019, 2024-07-19:
yea i tried both options and nothing happens.

sysadminpower2019, 2024-07-19:
[image]
Here is a screenshot from the serial option

richonguzman, 2024-07-19:
will try to write the YAAC dev to know what I should do.

Mane76, 2024-07-19:
Could you please try two variants, for both the TNC Option must be switched on in the iGate. I assume you run LINUX on your computer.

1) KISS over TCP: Enter IP from iGate, Port 8001, enter Callsign, enable Transmit, disable TEMP1-1, enable WIDE1-1, disable Beacon, Protocol RAW AX25

2) Serial Option: Device name /tmp/tnc, Bauf 115200, enter Callsign, enable Transmit, disable TEMP1-1, enable WIDE1-1, disable Beacon, Protocol RAW AX25
Then go to a console and enter  socat PTY,link=/tmp/tnc TCP:ip_of_iGate:8001 &

Mane76, 2024-07-19:
as 3rd option (but this is not your origin question), you can run XAstir as APRS Client with socat out of option 2:

[image]

sysadminpower2019, 2024-07-19:
it does seem that either the tcp/ip or the serial worked as i was receiving stations however when i try to send out any beacons nothing shows up on aprs.fi. The lora-aprs module does show up regularly sending beacons but nothing appears from YAAC.

Would the socat method then also theortically work for PAT?

sysadminpower2019, 2024-07-19:
at the moment. I am not entirely sure. there are not that many APRS stations around me to test if i am receiving and on which interface that is. I will need to try again

Mane76, 2024-07-19:
> it does seem that either the tcp/ip or the serial worked as i was receiving stations however when i try to send out any beacons nothing shows up on aprs.fi. The lora-aprs module does show up [...]

Take care not to mix up things:
When the iGate runs in TNC mode you can access it via Port 8001. This makes sense if either

- the iGate is not connected to APRS-IS and you want to see the stations on the map of XAstir or YAAC
- if the is or is not connected to APRS-IS and you want to send and receive messages through the iGates SSID via XAstir
- if the iGate is connected to APRS-IS and you additionally want to see the output

In all cases the functionality of connection to aprsis, digipeating, beaconing, ... shall be done via the iGate from Ricardo configurable in the webUI

Mane76, 2024-07-19:
> Would the socat method then also theortically work for PAT?

Basically yes, socat "emulates" a serial KISS interface to PAT and a TCP KISS interface to the iGate

sysadminpower2019, 2024-07-19:
yes, the lora-aprs module is transmitting correctly to aprs.fi the basic hardware installation works without any issues.

yes the ideal would be to be able to connect to the APRS-IS network via the hardware module, and then use YAAC or similar to send beacons, receive stations, send messages etc.

At the moment I am simply not sure if the TCP/IP is working or the Serial Variant is working, if anything is really working at all. 
All I can say is that at one point some additional station data did appear on YAAC. However I am certainly not able to transmit anything into the APRS-IS network from YAAC or similar

Mane76, 2024-07-19:
> yes, the lora-aprs module is transmitting correctly to aprs.fi the basic hardware installation works without any issues.
> 
ok
> yes the ideal would be to be able to connect to the APRS-IS network via the hardware module, and then use YAAC or similar to send beacons, receive stations, send messages etc.
> 
ok except the beaconing. This should be done through the iGates configuration

> At the moment I am simply not sure if the TCP/IP is working or the Serial Variant is working, if anything is really working at all. All I can say is that at one point some additional station data [...]

Socat and XAstir are working perfect with the configs I sent you, I have this running in test-setup

sysadminpower2019, 2024-07-19:
In terms of sending a beacon from YAAC. Would it not be possible to set the LORA board with callsign:5 and then send a beacon from YAAC with callsign:15

Mane76, 2024-07-19:
sure, but what is the sense behind ? 

Pure sending a beacon to RF or APRS-IS would work, but the way back (if someone wants to drop you a message would be broken up. 
For this case you would need to connect to APRS-IS via YAAC and use the iGate as pure hardware lora module. All "intelligence" would have to be done by YAAC. There are more elegant solutions.

Maybe you explain what you are targeting, then we can help you to find a suitable solution.

sysadminpower2019, 2024-07-19:
so if i am supposed to data transfered when using socat, I don't see anything. 
So there is still some issue

sysadminpower2019, 2024-07-19:
[image]

Mane76, 2024-07-19:
> so if i am supposed to data transfered when using socat, I don't see anything. So there is still some issue

How do you try to transfer data ?

sysadminpower2019, 2024-07-19:
I am not sure what you mean, which data exactly? isn't this simply supposed to show any data sent from YAAC when using the tnc option?

wb2osz, 2024-07-19:
I tried using YAAC with the LoRa Digi/IGate and it worked fine with TCP KISS.
Of course, you need to "Select APRS Server" in the LoRa Digi/IGate configuration.

Here is another approach to provide a bridge between LoRa APRS and the existing regular APRS RF network.
[https://github.com/wb2osz/direwolf-doc/blob/main/APRS-LoRa-VHF-APRS-Bridge.pdf ](https://github.com/wb2osz/direwolf-doc/blob/main/APRS-LoRa-VHF-APRS-Bridge.pdf )

73,
John WB2OSZ

sysadminpower2019, 2024-07-19:
then something may be wrong with my network @wb2osz becuase nothing seems to work as expected. 
the kissutil does seem to connect but i don't know much more after that.

Mane76, 2024-07-19:
> so if i am supposed to data transfered when using socat, I don't see anything. So there is still some issue

could you try to start socat with -v option ? Then it should show you more

sysadminpower2019, 2024-07-19:
nothing at all.
I even enabled the syslog option and there is only one entry 
APRSIS TX / StartUp_Status / https://github.com/richonguzman/LoRa_APRS_iGate 2024.06.15 
nothing more is being written to the logs. 
Sending and receiving packets to and from the module itself is fine and the module and it's SSID can be found on aprs-fi.

sysadminpower2019, 2024-07-19:
ah finally some data appeared on the socat -v output, although the syslog is still just one entry

Mane76, 2024-07-19:
ok, syslog is not working when on TNC mode, so this is normal.
When you RX or TX data it could be read in clear-text in the 1st digit of every line...

sysadminpower2019, 2024-07-19:
in any case the data that appears on the syslog output is only what is being sent from YAAC, nothing else. So the data is still not reaching the lora module for some reason.

Mane76, 2024-07-19:
basic question, in iGate config, have you:

- Section "Station", Symbol, set to "Red Diamond with L", entered valid callsign
- Section "APRS-IS", enable "Enable APRS-IS connection" and "Gate APRS-IS Messages to RF"
- Section "Beaconing", enable "Send our beacon to APRS-IS"
- Section "LoRa", enable "Enable LoRa-TX" and "Enable LoRa-RX" and set to same freq 

?

sysadminpower2019, 2024-07-19:
Yes, this is why the iGate works and send packages to aprs.fi without an issue. There is something else happening when trying to connect to the igate either via serial or tcp/ip which doesn't seem to work

Mane76, 2024-07-19:
what is the callsign and ssid which is on iGate and YAAC ?

sysadminpower2019, 2024-07-19:
i'd rather not provide my callsign, in any case the callsign with the iGate ssid is clearly visible on aprs.fi.

Mane76, 2024-07-19:
please have a look on https://www.findu.com/cgi-bin/raw.cgi?call=CALLSIGN and replace your callsign and ssid in this link. 

The link is RAW APRS_IS data.
Check if something was TX from the iGate which looks like what you wanted to have.

sysadminpower2019, 2024-07-19:
its interesting because now of all of a sudden the two ssid's from YAAC just appeared on APRS.fi. So I really don't know what is going on

closed 2025-01-24 by richonguzman