richonguzman/LoRa_APRS_iGate #140: Add Support / Documentation for WOAD on Android

sysadminpower2019, 2024-07-19:
It would be great if we could also use the lora board to connect to winlink on android via WOAD.

thank you

richonguzman, 2024-07-20:
Yes

But this is entering the twilight zone:

Ax.25 over B2F to use this 

WOAD and RadioMail use this and I've been trying for so long to make it work without any success

Do you know B2F?

sysadminpower2019, 2024-08-02:
Hola. Pudiste resolver el tema de WOAD? Lo marcaste como si ya estuviera hecho. Me supongo que quizas lo querias marcar que no le vas a poder hacer

richonguzman, 2024-08-02:
> Hola. Pudiste resolver el tema de WOAD? Lo marcaste como si ya estuviera hecho. Me supongo que quizas lo querias marcar que no le vas a poder hacer

Hola, como explique el WOAD usa el protocolo B2F y sin ayuda no tengo como seguir en eso ya que no pillo la forma para procesarlo. Me encantaría poder seguir en eso. Porque hace rato quiero un Gateway Winlink para lora, pero no logro dar con mas info

sysadminpower2019, 2024-08-04:
Hola, si gracias por la respuesta.
Te entiendo, Quizas voy a buscar informacion tambien para ver si te puedo ayudar con este tema.
En cualquier caso seria mejor dejar el ticket abierto por que puede ser que hayan personas interesadas tambien que puedan ayudar pero si el ticket esta cerrado entonces la gente no sabe

richonguzman, 2024-08-04:
ojala alguien se sume

hope someone know B2F to add this to LoRa world too