richonguzman/LoRa_APRS_iGate #141: Support for Lillygo Ver1 boards

raytechnical, 2024-07-21:
I have several LILYGO TTGO Ver1 boards which would could be utilised as  I Gates, would it be possible to add support for these please?

richonguzman, 2024-07-21:
Are these boards v1.6??

raytechnical, 2024-07-21:
The boards are version 1 boards exactly like this https://meshtastic.org/docs/hardware/devices/lora/?t-lora=v1 

The OLED pin outs  are different from the later versions of this board but the LoRa pin outs appear to be the same?

richonguzman, 2024-07-21:
So I will need your help to add this:

Can you provide the pinout list needed for this to work?

DonJaume, 2024-07-21:
Jugué con 2 placas 868 Mhz de este modelo (ahora no los tengo a mano). Los pines OLED se corresponden al modelo de pines del "HELTEC_V2".  Faltaría comprobar el pin de Batería para ver donde está.

richonguzman, 2024-07-21:
Pero lo demás funciona con heltec v2 ?

Porque no intentas con Lora32_915?

raytechnical, 2024-07-21:
Here is the pin out diagram:-
[image]

DonJaume, 2024-07-21:
> Pero lo demás funciona con heltec v2 ?
> 
> Porque no intentas con Lora32_915?

Buenas tardes,
De este modelo solo tengo uno para hacer pruebas (lo tenia escuchando satélites) y no tengo mas para 868 para poder testear al 100%.

Con "heltec-lora32-v2".
 - Inicia bien
 - Lectura de voltaje interno incorrecto, marca en el lcd 0.28v
- Al cambiar las frecuencias a 868 se bloquea en la pantalla de inicio.

Con "Lora32_915" y TGO_T_LORA32_V2_1_915
- Hay que cambiar la asignación de pines del LCD a 4, 15 y 16
- La lectura de voltaje sigue dando 0.28v
- No hay bloqueo al cambiar las frecuencias a 868 y puedo escuchar que emite las tramas con un sdr.

Sobre la lectura de la batería no encuentro información clara y he probado con todos los ADC por lo que imagino que esta placa no tiene divisor interno para la lectura.

richonguzman, 2024-07-23:
there is not much info about the pinouts of V1.0

sadly its not recommended to use with battery in meshtastic at all

raytechnical, 2024-07-23:
This is the pin configuration from the Meshtatsic Github https://github.com/meshtastic/firmware/blob/master/variants/tlora_v1/variant.h

I have never used a battery on these devices as per recommendations and have only been powered by the USB port.

raytechnical, 2024-10-05:
Hi,  
It has been a while since anyone posted, has anyone had any more thoughts on adding support for this board for  IGATE only, running off USB power supply?

Many thanks.

richonguzman, 2024-10-05:
> Hi, It has been a while since anyone posted, has anyone had any more thoughts on adding support for this board for IGATE only, running off USB power supply?
> 
> Many thanks.

could you help me out with all the pinout verification? 

I cant find any info about this board to add it

raytechnical, 2024-10-05:
Does this help?
LILYGO LORA32 V1.0 Details

ESP32 0.96 Freq 886/915 Mhz

RADIO_SCLK_PIN  5       // GPIO5    - SX1276 SCK
RADIO_MISO_PIN  19      // GPIO19   - SX1276 MISO
RADIO_MOSI_PIN  27      // GPIO27   - SX1276 MOSI
RADIO_CS_PIN    18      // GPIO18   - SX1276 CS
RADIO_RST_PIN   14      // GPIO14   - SX1276 RST
RADIO_BUSY_PIN  26      // GPIO26   - SX1276 IRQ

OLED Details

Type SSD 1306

OLED_SDA    4
OLED_SCL    15
OLED_RST    16

raytechnical, 2024-10-05:
It looks similar to the Heltec board.

raytechnical, 2024-10-06:
> It looks similar to the Heltec board.

Thats in respect to the pin outs.

raytechnical, 2024-10-06:
I take this issue is still open

raytechnical, 2024-10-08:
Hi All,
I have now managed to get the board working as an Igate, all appears to be working except I am unable to get the OLED display working.  Has anyone got any ideas why this does not work with this firmware, I am using ports
OLED_SDA 4
OLED_SCL 15
OLED_RST 16

Many thanks.

raytechnical, 2024-10-08:
I have added that to the display.cpp and re-flashed the board, unfortunately it is still the same, I must be missing something?  You are correct in your diagnosis as I have had a look at other coding with this board version and it shows a similar reset sequence for the OLED board.

richonguzman, 2024-10-08:
> I have added that to the display.cpp and re-flashed the board, unfortunately it is still the same, I must be missing something? You are correct in your diagnosis as I have had a look at other coding with this board version and it shows a similar reset sequence for the OLED board.

show please how you added this

DonJaume, 2024-10-08:
Buenas tardes,
Por si sirve de interés, a mi en la versión actual tampoco tampoco enciende la pantalla.
Con la build "2024-10-08" que tenía guardada de la última vez que lo probé, enciende bien. Solo cambié los pines.

[image]

raytechnical, 2024-10-08:
Still no further forward ..... will try again with a new build.

richonguzman, 2024-10-08:
> Buenas tardes, Por si sirve de interés, a mi en la versión actual tampoco enciende la pantalla. Con la build "2024-10-08" que tenía guardada de la última vez que lo probé, enciende bien. Solo cambié los pines.
> 
> ![foto](https://private-user-images.githubusercontent.com/31318417/374649106-59611488-21e0-412b-91ea-7c61a5a9cc65.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0MTI1OTgsIm5iZiI6MTcyODQxMjI5OCwicGF0aCI6Ii8zMTMxODQxNy8zNzQ2NDkxMDYtNTk2MTE0ODgtMjFlMC00MTJiLTkxZWEtN2M2MWE1YTljYzY1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA4VDE4MzEzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0NzFkN2Y2ZjIzZmIwNTM5OWZmYTFiZTI3ZGJmMGI0MDhhZGE5MWQ5MjNhN2Q2Nzc1MDhlYjYyMjk2YzRmMmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.s-gPSuL214JskYDFUB_6lEP9YOK4hyWEsFau6dms8v0)

cuentame, ahora si te funciono? para estar atento a hacer las modificaciones necesarias

raytechnical, 2024-10-09:
These are my modifications:-

 **platformio.ini**

default_envs = ttgo-lora32-v21_915

**Display.cpp**

ifdef HAS_DISPLAY
    #ifdef HAS_TFT
        #include <TFT_eSPI.h>

        TFT_eSPI tft = TFT_eSPI(); 

        #ifdef HELTEC_WIRELESS_TRACKER
            #define bigSizeFont     2.5
            #define smallSizeFont   1.5
            #define lineSpacing     12
        #endif
    #else
        #if HAS_EPAPER
            //
        #else
            #include <Adafruit_GFX.h>
            #include <Adafruit_SSD1306.h>
            #if defined(HELTEC_V3) || defined(HELTEC_WS) || defined(TGO_T_LORA32_V2_1_915)
                #define OLED_DISPLAY_HAS_RST_PIN
            #endif
            #ifdef HELTEC_WSL_V3_DISPLAY
                Adafruit_SSD1306 display(128, 64, &Wire1, OLED_RST);
            #else
                Adafruit_SSD1306 display(128, 64, &Wire, OLED_RST);
            #endif
        #endif

**boards_pinout.h**

    #endif// LORA MODULES
#if defined(TTGO_T_LORA32_V2_1) || defined(HELTEC_V2) || defined(ESP32_DIY_LoRa) || defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_0) || defined(TTGO_T_LORA32_V2_1_915) || defined(ESP32_DIY_LoRa_915) || defined(TTGO_T_Beam_V1_2_915) || defined(TTGO_T_Beam_V1_0_915)
    #define RADIO_SCLK_PIN  5       // GPIO5    - SX1278 SCK
    #define RADIO_MISO_PIN  19      // GPIO19   - SX1278 MISO
    #define RADIO_MOSI_PIN  27      // GPIO27   - SX1278 MOSI
    #define RADIO_CS_PIN    18      // GPIO18   - SX1278 CS ---> NSS
    #define RADIO_RST_PIN   14      // GPIO14   - SX1278 RST
    #define RADIO_BUSY_PIN  26      // GPIO26   - SX1278 IRQ ---->DIO0
#endif

// OLED 
#if defined(TTGO_T_LORA32_V2_1) || defined(ESP32_DIY_LoRa) || defined(ESP32_DIY_1W_LoRa) || defined(TTGO_T_Beam_V1_0) || defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_0_SX1268) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(OE5HWN_MeshCom) || defined(ESP32_DIY_LoRa_A7670) || defined(TTGO_T_LORA32_V2_1_915) || defined(ESP32_DIY_LoRa_915) || defined(TTGO_T_Beam_V1_0_915) || defined(TTGO_T_Beam_V1_2_915) || defined(ESP32_DIY_LoRa_A7670_915) || defined(ESP32_DIY_1W_LoRa_915) || defined(ESP32_DIY_1W_LoRa_LLCC68) || defined(ESP32_DIY_1W_LoRa_Mesh_V1_2)
    #define OLED_SDA    4       
    #define OLED_SCL    15
    #define OLED_RST    16      // Reset pin # (or -1 if sharing Arduino reset pin)
#endif

closed 2026-09-30 by richonguzman