richonguzman/LoRa_APRS_iGate #144: Fix for iGate never returning from Auto AP

S57PNX, 2024-07-29:
I have noticed that some of my gates drop off wifi and remain in Auto AP mode after some weeks of uptime, so I decided to do some tests. Analysis shows that gate never reconnects to wifi after an temporary outage of wifi, but it remains stuck in AP mode forever. Same thing was reported in #110 (gate booted before wifi router, so it entered Auto AP and never connected to the real wifi). In Monitor, I could see repeating message "Reconnecting to Wifi....", but no success. 

In line 43 of wifi_utils.cpp there is a call to function WiFi.reconnect() which is intended to reestablish the WiFi link to a previously configured SSID + password. In reality after returning from Auto AP mode, WiFi.reconnect() doesn't remember the previous SSID + password so it doesn't function. In this context, WiFi.begin () is a better call as it will apply the SSID + password from the config.

I decided to use a big hammer and replace WiFi.reconnect() with startWiFi() and this fixed the behaviour. I have tested various combinations of booting without wifi, removing and returning wifi signal, with and without backup digi functionality - it all works OK in my testing. Unfortunately I don't know how to make a pull request, but the fix is easy to apply manually.

Regards, Mitja

richonguzman, 2024-07-30:
hi! great info

please tell me which line you replaced exactly

"this" with "that"

to test it locally and add it to the main code

S57PNX, 2024-07-30:
In wifi_utils.cpp change this code 

`  if (!backUpDigiMode && (WiFi.status() != WL_CONNECTED) && ((millis() - previousWiFiMillis) >= 30 * 1000) && !WiFiAutoAPStarted) {
            Serial.print(millis());
            Serial.println("Reconnecting to WiFi...");
            WiFi.disconnect();
            WiFi.reconnect();
            previousWiFiMillis = millis();`

to

`  if (!backUpDigiMode && (WiFi.status() != WL_CONNECTED) && ((millis() - previousWiFiMillis) >= 30 * 1000) && !WiFiAutoAPStarted) {
            Serial.print(millis());
            Serial.println("Reconnecting to WiFi...");
            WiFi.disconnect();
            startWifi();
            previousWiFiMillis = millis();`

Change is in line 43.

1tilen, 2024-07-31:
I have tested the suggested fix, it looks like everything is woking as it is supposed to with S57PNX's fix.

richonguzman, 2024-08-02:
implemented!

closed 2024-08-02 by richonguzman