richonguzman/LoRa_APRS_iGate #145: iGate on SSID-11 won't beacon via KISS over IP set to same SSID as iGate

pine-marten, 2024-07-31:
running igate 2024.06.27 on a [ttgo Lora32 v1.6](https://www.lilygo.cc/products/lora3) 

default_envs = ttgo-lora32-v21

Works fine, will Tx a beacon if I press the send beacon now web page button from the web page. 
Device set to MYCALL-11 (using my own callsign not MYCALL ;)
APRS-IS is off, 
Enable TNC server is on

I am only looking at RF here. I Rx my signal from 32 miles away checking on aprs.fi. So far so good.

I connect to the TNC using Pinpoint APRS set to MYCALL-11. This seemed a logical thing to do, the unit and its antenna is MYCALL-11.

Will send messages via Pinpoint. iGate will NOT transmit a position beacon from PP. If I change the SSID in pinpoint to anything else the beacon is sent from PP (as well as from the web page, but on their respective SSIDs)

Switched to APRS-IS 32. I observe the same oddity - will not send position beacon if KISS program has the same SSID

Is this a bug or am I doing something that is wrong? Oddly enough with the [tracker ](https://github.com/richonguzman/LoRa_APRS_Tracker) 2024.07.01 I can set APRSdroid to the same call-SSID as the tracker and it's perfectly happy to send a beacon either by the standalone tracker or from APRSdroid Send Position, so this is inconsistent between devices

richonguzman, 2024-07-31:
maybe its the filter to avoid listening to yourself (same ssid ) when you hear a (repeated) beacon which was "like you sended"

pine-marten, 2024-07-31:
> ts the filter to avoid listening to yourself

This is purely RF Tx, it's not connected to APRS-IS. The iGate never Tx's in response to the pinpoint command over the KISS modem - the green light doesn't come on. And only the position beacon, I can send messages OK from Pinpoint APRS32

I'm happy to accept that using the same SSID is just a weird thing to try and do. It's not a killer issue, but I wasted some time trying to get it to work on the same SSID in testing. A warning saying don't use the same SSID on the KISS modem as the unit itself would save others the time finding that out ;)

Mane76, 2024-08-01:
Go to iGate WebUI and enable "accept own frames via KISS" in TNC section. Then it will work.

It definitly makes sense to have the same ssid in iGate and aprsdroid.

BR
Mane

pine-marten, 2024-08-01:
> Then it will work.

It does now. Sometimes, and I am not able to determine why in a repeatable fashion :( 

Thanks for confirming it wasn't barmy to set the same SSID. I've turned off the KISS modem now as these boards are cheap enough I'll make another tracker w/o GPS, and value this iGate for what it does well, being an iGate

closed 2024-08-01 by pine-marten