richonguzman/LoRa_APRS_iGate #146: [Feature Request] Option to send battery voltage as telemetry data

m1geo, 2024-08-02:
For adding an iGate in a remote-ish location, would it be possible to send the battery voltage as telemetry? This means that services like APRS.fi will graph the data over time, I understand it is sent in the comment field at present, but a telemetry packet would be great.

Thanks,
George, M1GEO.

richonguzman, 2024-08-02:
Hi George, 

we've been talking about it for a while:

it will need a second packet Tx for just sending the Battery Data, are you aware of it?

wb2osz, 2024-08-02:
There is another form of telemetry data which is embedded in a comment.
See   https://github.com/wb2osz/aprsspec/blob/main/APRS12b.pdf  near bottom of page 68.

73,
John WB2OSZ

richonguzman, 2024-08-02:
Hi John , thanks for the info: but the example is pretty confusing:

could I write you over telegram or signal to do some test and try to test it?

wb2osz, 2024-08-02:
Apparently telegram and signal are some sort of chat systems.
I don't know anything about them.
I'm still stuck in the e-mail age.
I will send e-mail with clarifications.

m1geo, 2024-08-02:
Hi @richonguzman

Sorry, I'm quite new to the project. I'll be honest, I haven't looked in detail at the code, so I'm not sure what else would need to be added. Telemetry packets aren't any more complex than the location/comment packets you already have, they just carry a different payload. [Some notes here](https://github.com/PhirePhly/aprs_notes/blob/master/telemetry_format.md).

I have access to some interesting towers but will need to run the iGates via Solar (no easy power on tower). I'm quite versed in solar power systems, but a way to monitor battery voltage (inferring battery health) from that is very useful. Having the ability to easily see a graph on APRS.fi is super helpful.

No worries if it's out of scope for the current development; just thought it would be a nice-to-have and wasn't aware that it was a popular topic! :) 

Best wishes from Cambridge,
George M1GEO.

g3cwi, 2024-08-24:
+1 for battery via telemetry. Very useful to monitor the performance of solar igates/digipeaters.

richonguzman, 2024-08-24:
hi @g3cwi and @m1geo , @wb2osz pointed me the way to add this to the same single gps packet of igate/digi but I havent done the testing as im finishing a personal job idea and next week I will endeavor into the web installer for the tracker

but sure enough I will return to this ;)

m1geo, 2024-08-24:
Thanks @richonguzman - it would be awesome to have it on the iGate as well as the tracker.

Keep up the awesome work! It's truly appreciated!

George. M1GEO.

m1geo, 2024-08-25:
Just to add, @richonguzman - it would be nice to be able to send the internal and external voltages. For me, I use an MPPT LiPo charger and use the external sense to observe the solar MPPT input and the external sense to observe the battery voltage. Thanks again!

IU5HKU, 2024-09-10:
Good idea, a graph can be useful in controlling what happens to battery/solar panels...ok, it's another packet and more energy consumption, but you don't need to send telemetry every minute, a packet every 5 or 10 minutes is enough, and you can use a switch in the gui to choose to send it or not...
Anyway, great piece of code, thank you so much!

richonguzman, 2024-09-11:
@wb2osz finally got it working and I'm testing it

sending packets at 10min rate got 2 ok and the somehow the next ones were ignored and only preocessed a 30min packet (and the rest at 10 min rate somehow got not processed)

IU5HKU, 2024-09-13:
Ok, so no less than 15 minutes between packets to be sure that every packet will be processed, the graph will be smooth enough.

richonguzman, 2024-09-13:
I got it working!! now I have to adapt it to the igate firmware, but its working even with fast data rate

m1geo, 2024-09-13:
Nicely done!

richonguzman, 2024-09-23:
thanks to @wb2osz info I got it working and was recently added for Release V2.0 on WebInstaller ;)

closed 2024-09-23 by richonguzman

m1geo, 2024-09-25:
With the configuration items as below, the iGate does not send any telemetry data; I have observed over the past 24h, and restarted it a couple of times to be sure.

[image]

richonguzman, 2024-09-25:
if the igate is sending Wx data , the telemetry cant be decoded from the comment part of the packet

please confirm the Blue+L symbol is active on your igate

m1geo, 2024-09-25:
Yes, the device is also sending weather telemetry with a BME280.

[image]

That explains it. Thanks.

richonguzman, 2024-09-25:
> Yes, the device is also sending weather telemetry with a BME280.
> 
> ![image](https://private-user-images.githubusercontent.com/758615/370712745-00511a48-8d21-49b3-a14f-3f6df42b758d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNjg0MzIsIm5iZiI6MTcyNzI2ODEzMiwicGF0aCI6Ii83NTg2MTUvMzcwNzEyNzQ1LTAwNTExYTQ4LThkMjEtNDliMy1hMTRmLTNmNmRmNDJiNzU4ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNVQxMjQyMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYjhiOWE5YzQwN2I1YzI5NzZmY2E1MTNhMTE4ZDE5MWQ1ZmQxOTFjMGVmMmU2ZGM5MTQ5Mjk2Y2E3NjM3Njg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.r2f3jJtENMrStnXYB_P7cB20syd4ChQIy1IMPrWYMQk)
> 
> That explains it. Thanks.
> 
> For anyone else, it appears you cannot have weather telemetry and voltage telemetry.

sadly I discovered this to, this is why I recommend to not having the "send encoded telemetry" switch on the station with Wx data

m1geo, 2024-09-25:
Fair enough. I have boldened the comment in my earlier post. Shame, but that's life. :)

iw2ejh, 2024-09-25:
Hope you can found a workaround as this will be very usefull for remote solar powered wx stations. Thanks