richonguzman/LoRa_APRS_iGate #149: Callsign does not pass validation.

Geksogengo, 2024-08-04:
My current callsign R1ZAAG is not validating, can this be fixed? Thanks a lot!

richonguzman, 2024-08-05:
hi 

is this a valid callsign?
as far as I know if callsign length is 6 characters the third position is the number

Geksogengo, 2024-08-05:
https://www.qrz.com/db/R1ZAAG 
4 category

richonguzman, 2024-08-05:
I never saw this type of callsign was a valid one

will try to validate it in code

Geksogengo, 2024-08-05:
[image]

wb2osz, 2024-08-05:
Here is a list of callsign prefixes:

https://www.arrl.org/international-call-sign-series

73,
John WB2OSZ

richonguzman, 2024-08-05:
@Geksogengo 

please test 2024.08.05 version :

it should have the fix to your callsign

Geksogengo, 2024-08-06:
[image]
[image]

richonguzman, 2024-08-06:
You need to test the new version of github (not web installer)
Web installer will be updated as soon as this is validated

Geksogengo, 2024-08-06:
[image]
[image]
Thank you, everything works! Many thanks for your project, 73!

richonguzman, 2024-08-06:
great!

richonguzman, 2024-08-06:
@Geksogengo remember tu use different ssid in your stations as you have at least 2 stations with ssid = -10

closed 2024-08-06 by richonguzman

Geksogengo, 2024-08-06:
> @Geksogengoпомните, что вы должны использовать разные ssid на своих станциях, так как у вас есть как минимум 2 станции с ssid = -10
I just recently discovered this error, it's strange that my colleagues didn't tell me about it earlier. But that will only happen in September, when I'll be home. Thanks!