richonguzman/LoRa_APRS_iGate #150: [Feature Request]Print WiFi MAC address at boot

m1geo, 2024-08-07:
Could the device print a MAC address on boot to the serial debug console. My office network incorporates MAC address filtering (actually, its about ownership), so it would be good to be able to find this easily.

Thanks!

richonguzman, 2024-08-07:
Hi,

do you have access to the router rules and current clients ?

this is just to know but why also MAC address of the board? what is filtered? 

like giving permission to boards with only certain MAC addresses as defining static internal net IP is not posible?

m1geo, 2024-08-07:
Hi. No. I do not. It is a corporate guest network. I can connect to the WiFi, but unless the device accepts the T&C or is sponsored by an employee (see below) then the device cannot access the internet.

In the end I used the ESPTool's `read_mac` instruction to get the MAC address, but I'm not sure of the WiFi stack you're using follows this MAC or if it generates a random one.

To sponsor a device, I just need to match the devices MAC address with my LDAP account - easily done, but I need to find the MAC address for the device. It would be cool if this was printed on the LCD for a second while connecting to WiFi or sent via the serial log.

Mane76, 2024-08-08:
As a workaround, you can connect the iGate to your local accesspoint of your mobile phone, then you see the used MAC.

m1geo, 2024-08-08:
Thanks @Mane76 - that's a good idea. I just don't want to do that at the office, as I'll confuse all the guest devices when they decide to jump to my phone as the RSSI is better! ;) The easiest option is to just print the MAC at boot. Hence the feature request. Most people will just ignore it, but it's useful for those who want it.

I did use the ESP32Tool to get the MAC but I am not sure that they use the same MAC.

```
C:\tmp\esptool-win64>esptool --chip esp32-s3 --port COM7 read_mac
esptool.py v4.7.0
Serial port COM7
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: 64:e8:33:69:00:04              <---  here
Uploading stub...
Running stub...
Stub running...
MAC: 64:e8:33:69:00:04              <---  here
Hard resetting via RTS pin...
C:\tmp\esptool-win64>
```

Mane76, 2024-08-08:
> s I'll confuse all the guest devices when they decide to jump to my phone as the RSSI is better!

Normally this should not happen when you give your phone a unique ssid and password and connect the iGate with this credentials.

I checked the esptool code with my iGate, here I have also two MACs shown and they are also equal. And with this MAC the iGate is connecting.

BR
Manfred

m1geo, 2024-08-08:
> this should not happen when you give your phone a unique ssid and password

Agreed; but since it does connect to the WiFi, the AutoAP doesn't start. I cannot change the SSID/PWD easily, so my easiest option is to spoof work's guest network - which is a bad idea for keeping my job ;)

The easiest option would be to print the MAC in future versions of the code during boot. Anyone who did need it would have it. Simple. And only requires half a line of C ;)

richonguzman, 2024-08-13:
Mac address printing over serial output in this new release (today)

closed 2024-08-13 by richonguzman