richonguzman/LoRa_APRS_iGate #152: Add support for Meshtastic-DIY board

Tapkin12ru, 2024-08-09:
Hello!
Can you add support for https://github.com/NanoVHF/Meshtastic-DIY/tree/main/PCB/ESP-32-devkit_EBYTE-E22/Mesh-v1.02-2LCD-FreePins board?
Pinouts for it in attach
[image]
Existing build ESP32_DIY_1W_LoRa in your project has another pinouts and not compatible with our boards... This causes problems with updating the firmware. Every time you have to remove the gate and update it locally with a build assembled with pinout changes.

richonguzman, 2024-08-09:
please write me the pinout definitions for your board to add it as a valid one

Tapkin12ru, 2024-08-09:
I attached correct pinouts for my board in 1st message. Please compare them with the esp_diy_1w_lora board_pinouts section in your firmware and you will see the difference

Tapkin12ru, 2024-08-09:
#define RADIO_SCLK_PIN  5
    #define RADIO_MISO_PIN  19
    #define RADIO_MOSI_PIN  27
    #define RADIO_CS_PIN    18
    #define RADIO_RST_PIN   23
    #define RADIO_DIO1_PIN  33
    #define RADIO_BUSY_PIN  32
    #define RADIO_RXEN      14
    #define RADIO_TXEN      13

richonguzman, 2024-08-12:
please test now:

ESP32_DIY_1W_LoRa_Mesh_V1_2

environment on new update to validate it and add it later to web-installer

Tapkin12ru, 2024-08-13:
Works fine!

richonguzman, 2024-08-13:
awesome, so it will stay ;)

closed 2024-08-13 by richonguzman