richonguzman/LoRa_APRS_iGate #154: Wheater data rainfall

TF1987, 2024-08-14:
Is there any chance to implement a rainfall sensor?

vy 73 de Thomas, DL8TF

richonguzman, 2024-08-14:
Hi @TF1987 , soon I will share a full WX station

meanwhile , because of the way the rain sensor works is somehow difficult to do it.

give me a few days more to share the Wx station over lora

closed 2024-08-14 by richonguzman

TF1987, 2024-08-14:
Hi, sounds nice. We had a big flood here beginning June because of heavy rain. And now it would be very interesting to log some rain data.

vy 73 de Thomas, DL8TF

richonguzman, 2024-08-14:
you would need LORA32
RS485 to TLL board
RS485 wind direction sensor
RS485 wind speed sensor
bme280
and also lux meter 
and also rain mm sensor

is that ok?

TF1987, 2024-08-15:
As mentioned, I will only need rain sensor. Do you have any idea what kind of sensors are suitable? The firmware should then be able to read the sensor and send the data to APRS-IS.

vy 73 Thomas

richonguzman, 2024-08-20:
take a look here :) 

https://github.com/richonguzman/LoRa_APRS_Wx_Station