richonguzman/LoRa_APRS_iGate #158: More issues with Wifi AutoAP

S57PNX, 2024-08-15:
In #144 we fixed the return from AutoAP mode to Wifi client mode. Unfortunately this uncovered a new issue that was hidden before: when AutoAP stops due to timeout, it correctly tries to connect to Wifi client and after  this fails, it will immediately start AutoAP again.

On a site without wifi, this means that AutoAP will be constantly on. It will timeout but restart immediately. This is convenient but wastes electricity on solar systems. 

My expectation is that AutoAP is only tried once after reboot - when it times out and there is no wifi present, it should remain off and save power while checking if wifi gets back. If there is no wifi and you need to service the gate, you need to reboot it to get into AutoAP mode.

In my mind the logic should be:
1. after reboot, try to connect to wifi
2. if wifi is not present, start AutoAP one time only until timeout
3.  after timeout, check if wifi is available
4. if no wifi, wait 15 minutes and go to 3.

This logic needs another wifi check loop and some refactoring of the code, which is a bit too much for me at this late hour in the evening. Better ideas?

richonguzman, 2024-08-15:
My plan is to stop all WiFi with button for new ultra low sleeping (until packet rx) digirepeater

S57PNX, 2024-08-15:
I just went through the code and I think I can make a minimal change to make autoap run only once, and then only try to connect to wifi evey 15 min. Would this help for now?

S57PNX, 2024-08-16:
I just created merge request #159 to fix this behaviour. While I have been testing this extensively in my lab with different configs, as usual more testing is recommended.

S57PNX, 2024-08-23:
I have been running this fix on two solar digi sites for a week and no problems observed so far. Reducing power consumption from 85mA to 37mA is a nice improvement, it makes my batteries last twice as long.

richonguzman, 2024-10-07:
added a WIFI KILL Switch (on Wifi AP section of webUI config)

this is to kill Wifi and WifiAP completely from restart for pure Remote Digirepeater  which should lower power comsumption a lot (as we need for this special cases)

please test it a comment

richonguzman, 2024-10-21:
Digi EcoMode not kills all and uses as low as ~21,5mA for esp32s3+sx1262 and ~24mA for esp32+sx1278 ;)