richonguzman/LoRa_APRS_iGate #160: Error when trying to OTA update

Tapkin12ru, 2024-08-17:
Hello.
I tried to update my station on TTGO from version 1.0.4.2 to version 1.0.5.1 via Ota update. 
[image]

When sending the file ota_update.bin I received this error. So it is impossible to update the firmware via Ota?
[image]

richonguzman, 2024-08-17:
hi,

which file did you upload?

Tapkin12ru, 2024-08-17:
This
[image]

richonguzman, 2024-08-17:
I always upload the firmware.bin file

will check what happened... can you flash completely fresh?

Tapkin12ru, 2024-08-17:
> I always upload the firmware.bin file

Did you mean this file?
[image]

> will check what happened... can you flash completely fresh?

I can't do it today because station installed outside of my QTH.

richonguzman, 2024-08-17:
yes this... please try it and let me know.

Tapkin12ru, 2024-08-17:
The same error:
"Upload failed
Could Not Activate The Firmware"

richonguzman, 2024-08-17:
but if you power it off and then on does it run?

Tapkin12ru, 2024-08-17:
I also tried to update via OTA for version 1.0.5
There is also an error, but a different one.
[image]

Tapkin12ru, 2024-08-17:
> but if you power it off and then on does it run?

Just now send command for reboot by UI. Station has rebooted and started without problems. But firmware is old.

richonguzman, 2024-08-17:
so at reboot works with old firmware as normal, but when trying the new firmware it does not work with OTA??

can you try again a few times to validate that its not a wifi speed or distance problem?

Tapkin12ru, 2024-08-17:
Now I took an identical TTGO board and flashed it via a web installer with erasing to version 1.0.4.2 and via OTA updated to version 1.0.5.1 without errors. Apparently, something bugged on my remotely installed station when I updated it from version 1.0.2 via OTA to 1.0.4.2. I will have to dismantle it and flash it cleanly to 1.0.5.1
And immediately another question: is it necessary to upload the spiffs.bin file to the board in addition to the firmware during OTA update?

richonguzman, 2024-08-17:
> Now I took an identical TTGO board and flashed it via a web installer with erasing to version 1.0.4.2 and via OTA updated to version 1.0.5.1 without errors. Apparently, something bugged on my [...]

no need to upload spiffs.bin

closed 2024-08-17 by Tapkin12ru