richonguzman/LoRa_APRS_iGate #162: Heltec Lora V3 entering sleep on boot with voltage present

pdevlin13, 2024-08-23:
I have been testing a Heltec Lora V3 running an iGate firmware and powered by USB. I have been monitoring the 12v battery supplying the USB buck convertor. The voltage has been reporting fine and so today I enabled monitor external voltage health, but it booted straight into sleep.
I am using default voltage divider values and pin 7 for the external voltage.
If I re-flash and restore the config with external voltage health turned off, the voltage is reading ok.

Any help appreciated
Paul
MW0PDV

richonguzman, 2024-08-23:
Paul

you were monitoring the 12V with the external voltage measurements and sending it over aprs, right?

 the battery being always over 10,9V, right?
 
 What has to do with the buck converter?
 
 what are your values for the voltage divider from the 12V battery?

pdevlin13, 2024-08-23:
Hi Ricardoyou were monitoring the 12V with the external voltage measurements and sending it over aprs, right?Yes external voltages have been sending to APRS fine. I have been testing as MW0PDV-15 for about a week.What has to do with the buck converter?Sorry was just trying to explain the 12 volt battery is being used to power ESP32 via a buck converter, I.e. no internal battery.the battery being always over 10,9V, right?Today I was testing on a bench power supply set at about 12.8 volts,  I have checked 15volts gives me 3v on pin 7.15volts on a bench supply reports 16.08 volts to APRS, see screen shot below from testing today.what are your values for the voltage divider from the 12V battery?The default 100k +27kThanksPaul.Sent from my iPad

richonguzman, 2024-08-23:
so, you've been monitoring the hole week the external voltage with good results

resistors and process works fine,

but as soon as you enable External Voltage Monitoring the boards turn off as somehow it does not see the correct value of the battery "health" and shuts down?

is the pin GPIO-7 ?

when you enable the monitor battery (is only external battery monitoring enabled and not internal battery monitoring?)

pdevlin13, 2024-08-23:
Hi RicardoThanks for quick response.Yes GPIO-7, the one highlighted below in screenshot.Also confirm I only have external monitoring enabled.Thanks Paul.Sent from my iPad

richonguzman, 2024-08-23:
when enabled it never boots again?

what values do you have on the low external monitor values? 10.9 ?

pdevlin13, 2024-08-23:
When saving the configuration after selecting monitor external battery, the displays goes blank. I have only waited 5 minutes but it just remains in sleep during that time.If I then use the web flash utility to connect and select monitor / console, do a reset, the console display  shows something like “sleeping low battery” The only way I know know to recover is to re flash the board.Is there a command to execute in the console to query external voltage or force out of sleep?I have 12.2v set as a sleep voltage, but I did try 3v to try and rule out errors in voltage calculation but this made no difference Paul Sent from my iPad

richonguzman, 2024-08-23:
this is really strange as you only monitor external value, which gives you correct values.

when re-booting , the board checks for the value (external in your case) and as I understand you validate its higher than "sleep voltage value".

so it should go on without any problem.

can you send me a picture on how you've set the switches on the BATTERY part of the WEB UI-configuration?

pdevlin13, 2024-08-23:
Below config will work and send external voltage to APRSAnd sent this report to APRSWhen the config below was saved it’s gone into deep sleepSent from my iPad

richonguzman, 2024-08-23:
when you say bellow I see no images added here

pdevlin13, 2024-08-23:
First config screen It’s APRS reportThen the config that sends the board into sleepSent from my iPad

richonguzman, 2024-08-23:
still no images

pdevlin13, 2024-08-23:
My iPad seemed to be stripping the images, I will send from my computer in the morning, I have tried sending with no annotation to see if that worksThanks Paul.Sent from my iPad

pdevlin13, 2024-08-24:
This is the config that works fine sending external voltage to APRS
[image]

pdevlin13, 2024-08-24:
This is the APRS report using the above config 
[image]

pdevlin13, 2024-08-24:
This is the config that when saved will put the board into sleep
[image]

pdevlin13, 2024-08-27:
Hi Ricardo, I tried doing some basic diagnostics using VS. If I inhibit the StartupBatteryHealth function from putting the Heltec V3 into sleep, I can see the checkExternalVoltage function is called several times but returns a value of 0.04, even with voltage on the GPIO pin. Eventually it returns the correct voltage. See attached screenshot of VS terminal window.
Paul.

[image]

richonguzman, 2024-08-27:
i was checking the code as I was suspicious that somehow the Config.battery.externalVoltagePin was not initialized before the batteryHealth function but all is working as it should.... so this may be a timing issue

maybe to add a delay to let the pin "really" read instead of just showing 0.04V???

test adding :

 delay(3000);
 
 
 at battery_utils.cpp , line 208 and move the code one line down
 
 maybe this delay could solve the issue or help us understand what's going on

pdevlin13, 2024-08-27:
Hi Ricardo

I tried the delay but that still didn't work.
Working through things a bit it seems that the variable voltageDividerTransformation isn't initialising properly. If I hard code a value of 4 instead as in batteryutils line 191 in the screenshot below, the board boots ok.
However it doesnt go into sleep but maybe thats because I polling APRS at 1 minute intervals for testing.

Paul.

[cid:5a008e72-4ebf-43a1-9611-3f56f8b5301b]

pdevlin13, 2024-08-27:
Hi Ricardo

I tried the delay but that still didn't work.
Working through things a bit it seems that the variable voltageDividerTransformation isn't initialising properly. If I hard code a value of 4 instead as in batteryutils line 191 in the screenshot below, the board boots ok.
However it doesnt go into sleep but maybe thats because I polling APRS at 1 minute intervals for testing.

Paul.
[image]

pdevlin13, 2024-08-28:
Hi Ricardo
I still need to change line 18 in Battery_utils to stop it going into sleep on boot.
It will now enter sleep on low voltage with the mod you have made

The line below works
[image]

This one sends the board to sleep
[image]

richonguzman, 2024-08-28:
new test , maybe this is the solution as I find the problem was the BATTERY_Utils::setup() was not called before doing the verification of batteryHealth at boot

so I add "BATTERY_Utils::setup();" in Power_Utils::setup() line 175, and deleted it from LoRa_APRS_iGate.cpp

pdevlin13, 2024-08-28:
Great thanks Ricardo, all seems to be working now :-) I will test over the next couple hours on MW0PDV-15, putting it in and out of sleep.

Really appreciate the work you are doing on LoRa Firmware/Software
Thanks
Paul.

richonguzman, 2024-08-28:
> Great thanks Ricardo, all seems to be working now :-) I will test over the next couple hours on MW0PDV-15, putting it in and out of sleep.
> 
> Really appreciate the work you are doing on LoRa Firmware/Software Thanks Paul.

good results!

thanks for the help!

let me know in a few days if everything is going fine to close the issue and rest on this ;)

pdevlin13, 2024-08-28:
Hi Ricardo
Testing update, the Heltec v3 board is now sleeping and waking up, thanks 😀
However there is now no beacon text before it enters sleep. Earlier I was seeing a Beacon text as below,
This is really useful for remote igates to know why they are offline 
[image]

richonguzman, 2024-08-28:
this beacon is sended when it was working fine and then the voltage got low

the warning does never happened when booting and the external voltage is still low

pdevlin13, 2024-08-28:
Hi Ricardo Ieft the board running, see example below, I have beacon interval set to 30min.
After yellow highlighted beacon at 19:55 I set the voltage low.
At 20:25 the board entered sleep as expected, but no beacon.
Once in sleep I set voltage high 
The green highlighted beacon is the wake up Beacon, having slept for 30 minutes 
This was working before the latest mod.

[image]

richonguzman, 2024-08-28:
updated code, I add code to force sending all packets of the output buffer as this maybe is losing the last packet that warns the soon to be sleeping board when Tx over RF (inserted of over Internet  / TCPIP)

please test

pdevlin13, 2024-08-28:
Brilliant thanks Ricardo, just done some quick testing it’s looking good😀
I will do some more extensive testing in the morning 
Thanks
Paul
[image]

pdevlin13, 2024-08-29:
Hi Ricardo, I have been testing for a few hours this morning, everything seems to be working fine with the external voltage monitoring 😀
Thanks again 
Paul.

richonguzman, 2024-08-29:
great!

closed 2024-08-29 by richonguzman