richonguzman/LoRa_APRS_iGate #168: Suggest adding iGate Digipeater symbol to UI

M0LTE, 2024-08-31:
I suggest adding iGate Digipeater symbol to the UI.

It is a green star with letter I.

```
"overlay": "I",
"symbol": "#",
```

This will be a common use-case.

S57PNX, 2024-08-31:
[image]

It has always been there in the UI. Or do you mean something different?

M0LTE, 2024-08-31:
According to https://github.com/PhirePhly/aprs_notes/blob/master/digipeater_symbol_selection.md and http://aprs.org/symbols/symbols-new.txt

Green star with I is "I-gate equipped digipeater" (I#)

Green star with L is "WIDEn-N with path length trapping" (L#)

My node is an I-gate equipped digipeater.

It works in the JSON config, just not in the UI.

S57PNX, 2024-08-31:
Ah, uppercase I vs. lowercase L makes your comment clearer. Per http://aprs.org/symbols/symbols-new.txt the following two symbols are used for gateways:

I& = Igate Generic (please use more specific overlay)
L& - Lora Igate

However, for digipeaters there is no analogy, becase L# is already used for something else.

I# - I-gate equipped digipeater
L# - WIDEn-N with path length trapping

The current usage of L# for an I-gate equipeed LoRa digipeater is logical, but the reference symbol table doesn't define a symbol for this combination, so you would like to fall back to the "generic digipeater with igate" symbol. Perhaps a new symbol should be requested for this from aprs.org ?

M0LTE, 2024-08-31:
Thank you for pointing out the LoRa-specific symbols/overlays. Since there is no perfect answer for this situation, I will make a local choice.

Up to you / the other contributors whether to adopt my suggestion or not, I am happy either way.

IMHO `L#` has no place in LoRa_APRS_iGate but it is present, however I'm not requesting its removal. `La` and `L_` also appear in the UI and appear to be non-standard, same applies.

S57PNX, 2024-08-31:
L_ is requred for aprs.fi to process the weather info, they trigger on blue color of the symbol. As a user, I would like to see these real-world use cases reflected back in the formal standard, but I don't know the process. Having said that, I think it is better to leave this decision to CA2RXU, it's his project.

M0LTE, 2024-08-31:
Absolutely agree.

I've customised my fork to work a bit more how I'd like it to. 

Adding the symbol I wanted:
[image]

Making it clear why `L_` is there, and showing the codes in the drop down list:
[image]

Adding a `?` symbol when there isn't an icon provided:
[image]

Feel free to adopt the commit, or not.

https://github.com/M0LTE/LoRa_APRS_iGate/commit/1d6adde

pdevlin13, 2024-08-31:
I think some caution is needed here, the standard for symbols on Aprs.org specifies and a Digipeater as a green star with white centre. It doesn’t seem to specify a preferred overlay for anything specific, hence different projects seem to have adopted their preferred overlay. There is a good practical argument for using L as the overlay for all things Lora. If I am looking for Lora Digi’s and/or igates L is very helpful. It also seems the L is already widely in use in the UK for Lora nodes