richonguzman/LoRa_APRS_iGate #171: Heltec Lora  32 V3 Display problem

pdevlin13, 2024-09-03:
Hi Ricardo
If I use the web installer to flash a board, the display on the Heltec Lora  32 V3 works ok.
If I use Visual Studio Code with the latest software/firmware from Github the display does not show. All other functions and serial messages work fine. Just wanted to raise this now to prevent any potential problem carrying through to the next release.

Thanks
Paul.

g3cwi, 2024-09-06:
I found that too today. I finally got my unit going with the web installer.

pdevlin13, 2024-09-06:
Hi Ricardo, I did have a go to try and work out what was going on. Strangely it seems to be calling the display class functions. I added some serial text, see below. I have added line numbers to  the text to make it a bit clearer See lines highlighted in yellow.

[image]

pdevlin13, 2024-09-06:
> I found that too today. I finally got my unit going with the web installer.

Hi Richard, although the display doesn't work, everything else seems to work fine. I just used APRS.fi to check my digipeater was up and running. 
Paul

pdevlin13, 2024-09-06:
Found a workaround :-) 
[image]
, After a bit of Googling on HeItec V3,I realised that the board SCL, SDA pins were being used to initialise the OLED. Changed the code below, adding or HETLEC_V3 and also changed pins to OLED_SDA, OLED_SCL instead of BOARD_I2C_SDA, BOARD_I2C_SCL

g3cwi, 2024-09-06:
If we can turn off the display and the wifi AP after a delay, that should reduce the average current and make battery/solar operation, a lot easier.

kecsa, 2024-09-09:
Hi,
In the new version (09.02), the display became dark.
I2C initialization was missing in Display.cpp.
After
#ifdef OLED_DISPLAY_HAS_RST_PIN ..........  #endif
I typed:
Wire.begin(OLED_SDA, OLED, SCL);
and that solved the problem.
Csaba

richonguzman, 2024-09-10:
@pdevlin13 , @kecsa , @S57PNX , @g3cwi please test latest update (MAIN BRANCH) which should correct the issue (versionDate = 2024.09.10)

kecsa, 2024-09-11:
Hi Ricardo,
I tested the 09.10. version.
It works great without any changes.
Thanks for the update and thanks for the great work.
Csaba.

g3cwi, 2024-09-12:
Tested. Working, thanks.

closed 2024-09-12 by richonguzman