richonguzman/LoRa_APRS_iGate #172: Heltec Lora 32 V3 Display Timeout

g3cwi, 2024-09-06:
Hoping to save power I set my display to switch off - but it remains on all the time.

S57PNX, 2024-09-06:
Not the solution to your problem, but to put some perspective on how much can we realistically gain by switching the OLED display off. Based on excellent measurements on https://bitbanksoftware.blogspot.com/2019/06/how-much-current-do-oled-displays-use.html ,  the 0.96" 128x64 monochrome OLED that we typically run on our Lora devices has an average consumption around 8mA at 3.3V.  This translates to approx 2mA on a 12V solar system. 

Is it really worth switching it off?

g3cwi, 2024-09-06:
>This translates to approx 2mA on a 12V solar system.

The display feels warm to the touch - but maybe that's components inderneath it?

S57PNX, 2024-09-06:
On your device the LoRa radio SX1262 is under the display.

richonguzman, 2024-09-10:
i made a few mods to the igate firmware to solve some HELTEC V3 oled not working

please test if you can.

g3cwi, 2024-09-11:
Tested the latest igate firmware. The display is not back on - thanks. The display and Wifi timeouts are not working at the moment.

richonguzman, 2024-10-07:
added a WIFI KILL Switch (on Wifi AP section of webUI config)

this is to kill Wifi and WifiAP completely from restart for pure Remote Digirepeater which should lower power comsumption a lot (as we need for this special cases)

please test it a comment

closed 2024-10-07 by richonguzman