richonguzman/LoRa_APRS_iGate #173: Feature request: Headless iGate (without OLED display)

btor, 2024-09-08:
Why:
1. Since many of our iGates are installed in a invisible place, e.g., near the antenna on the top of a mast, the display has no practical function.
2. Due to poor quality of the OLED display, after about 1.5 year 365/7/24 continuous use, the luminance of the display is severely reduced, and multiple pixels are failing.

Would it be possible to autodetect the presence of the OLED module during initialization and use it if present and change to headless mode if not?

richonguzman, 2024-09-08:
its a good idea ... I can try to check for it at boot up

but please hold on the idea as I'm dedicating this days to test and code for the tracker web installer 

please remember me about this in a week or two

Mane76, 2024-09-08:
maybe I am getting it wrong, what about to set in the config in section "display" the variable at "always on" to false and the timeout to e.g. 1 sec ? Then the display is off 99,99% of all time.

Just my 2 cents

steffenspies, 2024-09-08:
If I'm not mistaken, the Heltec Wirless Stick Lite V3 is already setup without using/setting a display.

Steffen / DL1SX