richonguzman/LoRa_APRS_iGate #174: HELTEC Wireless Stick No Display

9M2PJU, 2024-09-11:
Hello OM,
No display status on heltec wireless stick.

richonguzman, 2024-09-11:
we are currently testing as it should work and don't know why it does not... please share the exact model of your board

HELTEC WIRELESS STICK (which version? v2 or v3 ? or v3.1)?

9M2PJU, 2024-09-11:
> we are currently testing as it should work and don't know why it does not... please share the exact model of your board
> 
> HELTEC WIRELESS STICK (which version? v2 or v3 ? or v3.1)?

[image]

[image]

[image]

V3

richonguzman, 2024-09-12:
if seen the schematics of V2 and V3 

when using oled pins for V3 , wifi does not work and also oled does not light

curtosvienna, 2024-10-26:
No display status on heltec lora32 v2

downgrade to [V1.0.5.1] and now ok

richonguzman, 2025-01-10:
on HELTEC_V3_2 branch is a test firmware that should solve the issue of the Heltec Wireless Stick... please test it

closed 2025-01-10 by richonguzman