richonguzman/LoRa_APRS_iGate #175: External Voltage Measurement Heltec Wireless LoRa V3

g3cwi, 2024-09-12:
I have been trying to read an external battery Voltage - without success.

I tried GPIO 46 (board pin 5). That appears to be unassigned. No luck. I just get a constant Voltage reported that it not related to the actual voltage on the pin. I can change the reported values by changing the resistor values in the web interface but the reported voltage is not related to the actual applied voltage. I tried changing the actual applied volt (1.37-2.84 volts) but the reported voltage remains unchanged at 3.84 Volts. I rebooted after the changes in case it was only sampled infrequently but no change.

This would be a useful facility as my solar digi runs the board off a 5 Volt USB supply so I need to be able to monitor the batteries directly to understand their state of charge. Thanks.

richonguzman, 2024-09-12:
> This is on an iGate - not a tracker. Woops I have put the problem in the wrong section. Sorry.

no problem, but now we are sure we are talking about igate firmware:

test GPIO 6 or GPIO7

richonguzman, 2024-09-12:
remember to use voltage divider for voltages higher than 3.3v on the input pin

richonguzman, 2024-09-12:
great!

closed 2024-09-12 by richonguzman