richonguzman/LoRa_APRS_iGate #176: Calculate a proper IAQ when BME680 is in use

IU5HKU, 2024-09-13:
This isn't a proper issue, but a proposal for a better use of the "gas resistence" value obtained by BME680 reading.
This piece of hardware can sense the VOC (Volatile Organic Compounds) in the air, and using their proprietary library you can obtain a lot of interesting infos, but the library is way too big to insert here, and is closed sourced too (Bosch "bsec" library if you are interested, here https://github.com/boschsensortec/BSEC-Arduino-library).
Anyway, there is a smart piece of code written by David Bird (G6EJD) who calcs a IAQ (Index of Air Quality) using a short formula that i think can be used in our case. His repo here on github is here:

https://github.com/G6EJD/BME680-Example/blob/master/ESP32_bme680_CC_demo_03.ino

and as you can see is well explained and easily embeddable into the IGate code. Not precise as the bsec one, but precise enough and better than a raw ohmic resistance value.
There is an issue: for a proper measurement first the resistence must be heated for few milliseconds, ok, it consumes a bit of your battery, but you obtain a real value and not a random one....i think that it worth the battery consumption, and many of us have the iGate over the desk, with a proper power source...and aprs.fi can show IAQ in a graph, i think.
You must ask him for permission to include part of his code in your project, but i hope that he don't disagree, it's for a good cause... :-)
Hope this idea can be appreciated, i can help you in integration of this feature if you want, let me know.
Have a good day

richonguzman, 2024-09-13:
if you can help contacting him and making a small proposal to start with it, it would be awesome help

IU5HKU, 2024-09-13:
Sure, i'll write an email to David asking for the permission.

IU5HKU, 2024-09-14:
Hi Ricardo, David grants us the permission to use his code, he wrote me "Please go ahead and use the code." :D
As previous said, this is an approximation of the real IAQ (IAQ stands for Indoor Air Quality and not Index of Air Quality, sorry, my fault) but for sure is more useful compared to the raw ohmic value.