richonguzman/LoRa_APRS_iGate #180: voltage as telemetry - logic bugs

S57PNX, 2024-09-25:
I've noted some logical issues with the new functionality of sending voltage as telemetry (version 2024.9.23)

- the initial state of this function is enabled, even though the web config switch is off. Afterwards, toggling the switch works OK and disables the function on and off as expected. However, the initial state should be disabled. From a quick glance I think this is line 181 of configuration.cpp, where the state of battery.sendVoltageAsTelemetry should be false.

- it looks like aprs.fi does not accept telemetry packets if my gate's symbol is blue circle with L. This is not documented on aprs.fi and I can see some Analog APRS gates with blue symbol and telemetry, so I don't understand why this problem. Maybe other products  switch away from blue symbol for telemetry and then switch back to blue for the WX packets? 

I am traveling this week, so I will be only able to investigate more next week.

richonguzman, 2024-09-25:
- line 181 is true if no data can be read from igate_conf.json file. If found the value is the one on the json file (does this answer the first issue?)
- aprs.fi and no aprs pages decodes telemetry from a WX packet, so having blue + L as symbols and sending Wx telemetry packet won't allow the encoded telemetry to work.

I guess the way other stations send telemetry is by sending more messages in the other telemetry mode. My encoded telemetry add was to keep on sending one packet as normal procedure instead of 2 for just sending gps beacon and telemetry

S57PNX, 2024-09-25:
Yes, if the data is not present in igate_conf.json, line 181 is true while the switch in UI is off, this is not correct behaviour and is confusing for user. The default in 181 should be false because the switch default is false. Once the the value is written in the json file then the UI switch and software logic will follow the value from json.

Regarding 2, are you saying that voltage telemetry and WX are XOR ? What if somebody wants both? 

For me, tracking voltage as telemetry is more important on mountain sites (with solar power) than at home, and tracking WX is also more interesting on mountains.

richonguzman, 2024-09-25:
the thing is APRS webpages dont decode the comment part telemetry when packet is the Wx packet.

so for stations sending Wx packets I will add a second packet with just pure battery telemetry soon.

sadly I will have to send two packets , unless I send a telemetry packet once each our and not at each gps beacon Tx.

S57PNX, 2024-09-25:
Can you send beacon + telemetry in one packet, and wx in second packet?

or must be 3 packets - one for beacon, one for voltage and one for wx?

richonguzman, 2024-09-25:
normal telemetry works like this:
1 ) send 3 packets one single time that defines the data sended on telemetry
2 ) telemetry packet (not encoded at all)
3 ) send a normal gps beacon packet (which could be Wx packet or not)

encoded telemetry:
1 ) send 3 packets one single time that defines the data sended on telemetry
2 ) send a normal gps beacon packet with encoded telemetry added in comment (it must be NOT Wx packet)
or
2 ) send a normal gps beacon packet with Wx
3 ) send normal Telemetry packet with required data

S57PNX, 2024-09-25:
I see now, thanks. Can you do: 

1) send 3 packets one single time that defines the data sent on telemetry
2) send a normal gps beacon packet with encoded telemetry added in comment (it must be NOT Wx packet)
3) send a wx packet if wx is enabled

I see more people with telemetry enabled than with WX. So it makes sense to put telemetry in beacon and only send second packet if they have wx enabled.

richonguzman, 2024-09-25:
> I see now, thanks. Can you do:
> 
> 1. send 3 packets one single time that defines the data sent on telemetry
> 2. send a normal gps beacon packet with encoded telemetry added in comment (it must be NOT Wx packet)
> 3. send a wx packet if wx is enabled
> 
> I see more people with voltage measurement enabled than with WX. So it makes sense to put telemetry in beacon and only send second packet if they have wx enabled.

this would make 2 gps packets (one with encoded telemetry and one with wx data)

S57PNX, 2024-09-25:
Is sending 2 gps packets a bad thing?

I just did a quick statistics for my country (S5): we have 52 igates/digis in total. 

- 8 (15%) have WX enabled
- 20 (38%) have voltage enabled
- 5 (10%) have voltage and WX

So we would be sending 2 packets only from 10% of sites, as opposed to sending 2 packets from 38% of sites if telemetry is second packet.

Anyway, I don't feel strongly about this and leave the decision to more experienced people.