richonguzman/LoRa_APRS_iGate #181: Disable the GPS module on ttgo-t-beam-v1

MihaiRF, 2024-09-30:
Hello,
Just for consumption reasons, it would be good to have the possibility to disable the GPS on T-beam boards if they are used as fixed iGates.
I measured ~110mA consumption instead of ~50mA on a Lora32 board with no GPS. 
I would like to use a T-beam board as a solar powered digipeater and the consumption is significantly higher when GPS is enabled.
Could this be implemented? :)
Thank you.

Regards,
YO5OMT, Mihai

richonguzman, 2024-09-30:
Hi,

GPS was never activated , nor even defined in iGate FW

MihaiRF, 2024-09-30:
OK, thanks for clarification. 
I'll dig into this more when time allows. Maybe the board is faulty or something. I've seen a GPS_VDD line coming from the power chip and I'm curious if it gets power and draws any current there.
Thanks.

richonguzman, 2024-09-30:
Let me know what you find

S57PNX, 2024-10-01:
Mihai, check #158 and check if perhaps you are encountering this behaviour. It could be the reason for you 110mA consumption.

MihaiRF, 2024-10-01:
> Mihai, check #158 and check if perhaps you are encountering this behaviour. It could be the reason for your 110mA consumption.
The high consumption could be due to the AP not going off. I've noticed that no mater what I set in the "Auto AP power off timeout", the access point remains on. If it's a remote solar powered node, it makes no sense to keep it active.

S57PNX, 2024-10-02:
Small but important correction - AP is going off as expected, but due to described flaw it immediately restarts again if there is no wifi present, and will keep on restarting forever. My proposed solution is to run AP only once after reboot, so it will not keep on restarting. But this can be resolved in other ways as well.

MihaiRF, 2024-10-03:
Hi, 
I've just tested the consumption on a Lora32 board and it's around 150mA (probably due to AP being on). I get there are ways to resolve this but I don't have enough experience to make my own modifications in the source code. 
Could you implement a solution for this in the next release? I know that just the processor plus LoRa modem should talke about 40mA@3.7V to run. 
Thank you!

richonguzman, 2024-10-07:
I got news for all,

added a WIFI KILL Switch (on Wifi AP section of webUI config)

this is to kill Wifi and WifiAP completely from restart for pure Remote Digirepeater  which should lower power comsumption a lot (as we need for this special cases)

please test it a comment

PS: @S57PNX WiFiAP timeout restart is to do a few things, but mainly to check if Wifi is available again connect to it, or start again Wifi AP if needed, but this new method kills it all ;)

S57PNX, 2024-10-07:
I am not sure that this covers adequately my use case. 

My use case is that I have multiple solar powered gates that have wifi available, so I can't use wifi kill switch.  Being connected to wifi doesn't cost much in terms of power and it is worth the little extra power usage, so for me this an acceptable tradeoff. But sometimes the wifi goes away (from a few minutes to a week!), and wifiAP runs all the time with a big power consumption cost. 

If the wifi kill switch is off, can you still consider starting WifiAP only one time while still checking for wifi every 15 minutes? 
Checking if wifi is available again doesn't cost much in terms of power use, but starting wifiAP for 15 minutes costs a lot.

Finally, if wifi kill switch is on, how can the digi be serviced when you visit the site? The digi will not have wifi or wifiap. Do you need to bring a laptop with visual studio code and delete the configuration from the device?

WIth "my" solution you restat the digi and get wifiap one time for service, and then after timeout it goes away until next reboot.

MihaiRF, 2024-10-08:
Hi. 
I've tested the WIFI KILL switch and indeed, the consumption drops to ~50mA. However, I don't think that loosing control of the board completely is a way to go. I think starting AP once on each boot (for a few minutes) would still be a required thing to do if you still want to access the board and change some configuration. I think @S57PNX solution is reasonable and would do the job...

SQ2CPA, 2024-10-08:
Use ESP32C3+SX1262 and with low power mode (implemented in this software) you can get less than 3.3V 10mA power usage (like 7-8mA). 50mA is still too high, for that you need lot of PV power and energy storage so the station will probably fall in the winter

MihaiRF, 2024-10-08:
> Use ESP32C3+SX1262 and with low power mode (implemented in this software) you can get less than 3.3V 10mA power usage (like 7-8mA). 50mA is still too high, for that you need lot of PV power and energy storage so the station will probably fall in the winter

Thanks for the info. Could you provide an example for a board containing this configuration plus everything else needed to run? (2.4Ghz antenna, SMA connector for 433Mhz, display, etc). Thanks.

SQ2CPA, 2024-10-08:
I didn't checked that but heltec wireless stick lite have ESP32C3+SX1262 but I don't know what is voltage converter power usage. With display please don't be funny :) That's low power station and any mA matters here so we don't want any extra components, even voltage converter, MPPT effectivity must be good to not waste power.

I'm using HT-CT62 that you already mention. You have env in that software for that and I'm using it. You need VCC, EN (to VCC), GND, TX, RX and GPIO9 for programming (low during startup to enter flash). I implemented low power mode in that software specially for HT-CT62

MihaiRF, 2024-10-08:
> I'm using HT-CT62 that you already mention. You have env in that software for that and I'm using it. You need VCC, EN (to VCC), GND, TX, RX and GPIO9 for programming (low during startup to enter flash). I implemented low power mode in that software specially for HT-CT62

Sounds like a good choice then. The price is also nice. I think I'll order one and give it a try...

closed 2024-10-21 by richonguzman