richonguzman/LoRa_APRS_iGate #184: Feature request: Headless iGate (without OLED display) #173

btor, 2024-10-14:
Hi Ricardo,

You asked me to remind you after the tracker web flasher is completed.
See #173 opened on Sep 8

Best regards,
Bela / HB9HEY

richonguzman, 2024-10-14:
hi ,

please remember me: was it to have the igate board as digi or as igate without the oled screen?
which board was it?

btor, 2024-10-15:
Hi Ricardo,

Yes, it was the iGate/digi board. (I use the TTGO Lilygo LoRa32 v2.1 / v1.6  boards for iGates.)
Since I and many other users install them in a waterproof enclosure near the antenna on an antenna tower, the OLED display has no purpose.

richonguzman, 2024-10-21:
DigiEcoMode waas added in Release V2.0.1 which will disable Wifi, wifiAP , screen and leds and even lower CPU speed to use the least amount of current.

closed 2024-10-21 by richonguzman