richonguzman/LoRa_APRS_iGate #186: No more connect to APRS-server possible

dl7ata, 2024-10-23:
After udating to Version 2024-10-21 there is no connect possible to aprs server:
_Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 19
[ 30494][E][WiFiClient.cpp:275] connect(): socket error on fd 50, errno: 104, "Connection reset by peer"
Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 20_   ... running in a loop.

Did you change any library?

btw - thanks for this nice piece of software
73, Frank

richonguzman, 2024-10-23:
no library changes at all...  is your wifi ok? passcode ok? try another server?

dl7ata, 2024-10-23:
WiFi ok: I have access to the system via WiFi/LAN
Passcode ok: Nothing changed since previous version from Oct.
Server: rotate.aprs2.net is a rotating system which I can connect i.e. with aprx or other aprs - systems

Oled shows "normal" traffic, message comes via serial output

richonguzman, 2024-10-23:
which boards is this?

dl7ata, 2024-10-23:
T-Beam Meshtastic
Chip is ESP32-D0WDQ6-V3 (revision 3)
Meshtastic T-Beam ESP32 LoRa 433Mhz
SX1276/SX1278 Wireless Module WiFi GPS NEO-6M With 0.96 inch OLED Display For Arduino.

dl7ata, 2024-10-23:
btw: This is my hardware
[https://www.lilygo.cc/products/t-beam-v1-1-esp32-lora-module](url)

dl7ata, 2024-10-23:
Ok, will do and let you know.
Thanks!

dl7ata, 2024-10-23:
Flashed with webflasher, it's a V 1.2 (label on the top).
Same reaction:

_Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 7
[ 18031][E][WiFiClient.cpp:275] connect(): socket error on fd 50, errno: 104, "Connection reset by peer"
Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 8
[ 19050][E][WiFiClient.cpp:275] connect(): socket error on fd 50, errno: 104, "Connection reset by peer"
Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 9
[ 20067][E][WiFiClient.cpp:275] connect(): socket error on fd 50, errno: 104, "Connection reset by peer"
Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 10
[ 21085][E][WiFiClient.cpp:275] connect(): socket error on fd 50, errno: 104, "Connection reset by peer"
Didn't connect with server..._

richonguzman, 2024-10-23:
somehow aprs-is servers dont let you connect to it:

passcode calculated as per callsign info?
is it connected to wifi?

dl7ata, 2024-10-23:
I double checked my router - IP is not blocked into internet.
passcode is the same than for Digi DB0TGO (same callsign)
Connected to WiFi, I managed to configure it.

richonguzman, 2024-10-23:
does the two stations have the same Callsign with no SSID?

dl7ata, 2024-10-23:
No. The "Master-Digi"/aprs runs as ..TGO-14 and ...TGO-15.
But this could be the failure, a second login to the server. I think the server didn't recognize the SSID....
I will change the callsign to my personal one and will test it

dl7ata, 2024-10-23:
Same result:
_[ 17097][E][WiFiClient.cpp:275] connect(): socket error on fd 49, errno: 104, "Connection reset by peer"
Didn't connect with server...
Run client.stop
Trying to connect with Server: rotate.aprs2.net AprsServerPort: 14580
Try: 8
[ 18115][E][WiFiClient.cpp:275] connect(): socket error on fd 49, errno: 104, "Connection reset by peer"

richonguzman, 2024-10-23:
ssid does not matter, I have currently 4 different SSID (same Callsign) connected to aprs-is servers.

so there must be another problem.

So Wifi is connected and only the aprs-is connection is not completed.

how close are you to the router? any fixed IP on them that could block this?

dl7ata, 2024-10-23:
Router distance is 2m. The access via web to the T-Beam is excellent.
No fixed IP, DHCP is running and browser access is really fast for updating parameter or look up at received packets tab.

richonguzman, 2024-10-23:
any spaces In callsign declaration? please paste the configuration.json you save on the WebUI (minus relevant info).

dl7ata, 2024-10-23:
config.json:

> {"wifi":{"AP":[{"ssid":"##???##","password":"password"}],"autoAP":{"password":"abcd","timeout":10}},"callsign":"DL7ATA-7","aprs_is":{"active":true,"passcode":"onetwo","server":"rotate.aprs2.net","port":14580,"filter":"m/50","messagesToRF":true,"objectsToRF":false},"beacon":{"comment":"144.8/432.5/433.775MHz","interval":29,"latitude":52.5692293,"longitude":13.2312943,"overlay":"L","symbol":"#","sendViaAPRSIS":true,"sendViaRF":true,"path":"WIDE1-1","gpsActive":true,"gpsAmbiguity":false},"digi":{"mode":2,"ecoMode":false},"lora":{"rxFreq":433775000,"txFreq":433775000,"spreadingFactor":12,"signalBandwidth":125000,"codingRate4":5,"power":20,"txActive":true,"rxActive":true},"display":{"alwaysOn":true,"timeout":4,"turn180":false},"battery":{"sendInternalVoltage":false,"monitorInternalVoltage":false,"internalSleepVoltage":2.900000095,"sendExternalVoltage":false,"externalVoltagePin":34,"monitorExternalVoltage":false,"externalSleepVoltage":10.89999962,"voltageDividerR1":100,"voltageDividerR2":27,"sendVoltageAsTelemetry":true},"wxsensor":{"active":false,"heightCorrection":0,"temperatureCorrection":0},"syslog":{"active":false,"server":"192.168.0.100","port":514},"tnc":{"enableServer":true,"enableSerial":true,"acceptOwn":true},"other":{"rebootMode":false,"rebootModeTime":6,"rememberStationTime":30,"backupDigiMode":false,"lowPowerMode":false,"lowVoltageCutOff":0},"ota":{"username":"","password":""},"personalNote":"2beerOrNot2beer","webadmin":{"active":false,"username":"admin","password":""},"ntp":{"gmtCorrection":1}}

richonguzman, 2024-10-23:
write me on telegram at the same user please ... 

all looks good and should work as is..

dl7ata, 2024-10-28:
Based on a mismatch at  my router, it was not possible to connect. Hard to remove. It looks like a double entry. But with logon into my guest net, I could analyze and repair it.

closed 2024-10-28 by dl7ata