richonguzman/LoRa_APRS_iGate #188: Question: which format/encoding is used for sending data on LoRA?

bettio, 2024-10-26:
It is not clear how data is encoded before being sent over LoRa. Is any document format used?
What `0x3C, 0xFF, 0x1` / `0x3C, 0xFF, 0x17` strings mean, and what is the meaning of what follows after?

richonguzman, 2024-10-26:
from the first LoRa APRS firmware they uses 3 identification bytes to recognize a valid APRS LoRa packet

bettio, 2024-10-27:
@richonguzman thanks, still I'm not understanding how is encoded all the following bytes.
Do you have any link to any protocol / format documentation that you've been implementing?

richonguzman, 2024-10-27:
all of the lora packets is the APRS protocol written in plain text (or as TNC2) 

do you know [APRS101.pdf](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=http://www.aprs.org/doc/APRS101.PDF&ved=2ahUKEwj14NXAya6JAxVjqZUCHUxSL9gQFnoECBYQAQ&usg=AOvVaw0L7tq8UFcBMNQFgsrg6eGe)

wb2osz, 2024-10-27:
APRS101.pdf was written in 2000.  It is missing all of the corrections, clarifications, and new features since that time.
You can find newer information here:  https://github.com/wb2osz/aprsspec 
APRS12b.pdf is version 1.2 draft b.  Look for draft c soon.
A much gentler introduction can be found in "Understanding APRS Packets."

73,
John WB2OSZ

richonguzman, 2024-10-27:
> APRS101.pdf was written in 2000. It is missing all of the corrections, clarifications, and new features since that time. You can find newer information here: https://github.com/wb2osz/aprsspec [...]

you are right @wb2osz  I will forward all to this 1.2b version you have from now. :)

bettio, 2024-10-27:
I'm pretty confused: the following 3 bytes after the identification header have decimal values `73, 65, 219` (so the captured packet starts with ` 0x3c, 0xff, 0x01, 0x49, 0x41, 0xdb, ...`), and there are more and more characters that are out of the plain text range. Should I assume the packet is corrupted? or am I missing some decoding step?

richonguzman, 2024-10-27:
I guess you should explain what are you decoding and how to understand how you should read it

bettio, 2024-10-27:
I did a LoRa packet capture nearby a device running LoRa_APRS_iGate and I was trying to understand the packet format, but I wasn't understanding how to manually decode it.

So my plan right now is just to understand the format, and being able to manually decode it.

What I captured while the device was sending a beacon is:
```
    0x3c, 0xff, 0x01, 0x49,
    0x41, 0xdb,
[... ommited ...]
    0x91, 0x94, 0x35
    ```

It doesn't look plain text, and I'm not understanding the used format.

richonguzman, 2024-10-27:
are you printing them  (each byte) as HEX??? (try it printing as they are... just text)

bettio, 2024-10-27:
As text there are several unprintable bytes: 
[image]

Should I assume this is corrupt?

richonguzman, 2024-10-27:
i have now more questions than answers for this:

where did you read/hear this?
is it really APRS LORA or another lora message (maybe even encrypted)?
what is the program you use to hear/decode?
what are the settings you are using for decoding/listening. (lora settings?)

bettio, 2024-10-27:
> where did you read/hear this?

I captured this placing a lilygo device close to a friend lora transmitter running LoRa_APRS_iGate.

> is it really APRS LORA or another lora message (maybe even encrypted)?

The device transmitting this is running  LoRa_APRS_iGate.

> what is the program you use to hear/decode?

A custom firmware I did that prints to terminal lora packets

what are the settings you are using for decoding/listening. (lora settings?)

Just printing bytes in decimal as soon as a packet is received. Settings are:
```
      tx_power: 21,
      frequency: 433_775_000,
      bandwidth: :bw_125khz,
      spreading_factor: 12,
      coding_rate: :cr_4_8,
      preamble_length: 8,
      sync_word: 0x12,
      header_mode: :explicit,
      invert_iq: false,
      enable_crc: true
```

richonguzman, 2024-10-27:
try codingRate4: 5 instead of 8 (as this is the one used in 433.775 / 12 /125.000 / 5

i dont know about the sync word too, I remember it was 0x34 as the typical default sync word but you would need to dig a little on google to make sure of it