richonguzman/LoRa_APRS_iGate #193: Internal voltage level controll

RA3TOX, 2024-11-17:
Tell me, how is the internal voltage level controlled?
I enabled the monitoring option, but it goes into sleep mode. The internal voltage shows 0.

My board is LILYGO®TTGO LoRa32 V2.1_1,6 433 МГц and is powered by a 3.6 V battery through a standard connector.
Now I have no access to the device and cannot reprogram it.

What should I do?

richonguzman, 2024-11-17:
pin35 is the internal battery monitoring pin. 

you should validate that the battery is read correctly on the board before you put the monitoring process. 

if it works Fine, the monitoring process should validate that the voltage is higher than the SleepVoltage and if lower it will sleep

so first: check that the board reads the battery voltage

RA3TOX, 2024-11-18:
How to properly connect voltage to contact 35?

RA3TOX, 2024-11-18:
[image]
Is this correct?
What are the resistor values?

S57PNX, 2024-11-18:
You don't need to do anything. The internal voltage is internally connected on the board, the voltage divider is already on the board and the measurement should work without any modifications.

As @richonguzman  says, disable voltage monitoring for now for easier troubleshooting until you get a valid voltage reading. Enable "Send internal voltage" and observe if the screen shows 3.6V or similar. If it shows 0V, you have two options:

a) you didn't flash the correct firmware for this board and the power management chip is not detected
b) your board is broken. Measure the voltage between PIN 35 and GND with a voltmeter and see if the voltage is non-zero.

The external voltage divider is only for external voltage measurements via pin 34.

RA3TOX, 2024-11-18:
At pin 35 this function does not work ;(
[image]

S57PNX, 2024-11-18:
Try to monitor USB during boot (using VS Code) and see if the power management chip is detected correctly. (my point a) above).,

S57PNX, 2024-11-18:
I just took a fresh Lora32 2.1 out of the box, flashed it with Lora32 2.1/1.6.1 firmware and enabled "Send internal Voltage". I can see the voltage on screen and on pin35 (lowered by the onboard internal voltage divider).

[image]

[image]

RA3TOX, 2024-11-18:
@S57PNX   What is your firmware date?

S57PNX, 2024-11-18:
For this example I used the latest firmware, 2024.11.6. But any other firmware from earlier this year should work, the voltage measurement code is old and hasn't changed in a long time.

RA3TOX, 2024-11-18:
@S57PNX and still this function does not work correctly ...
How to set the correct voltage?

S57PNX, 2024-11-18:
If you measure 0V on pin35, it is not a matter of firmware. It means that your board is somehow broken. Try with another board if you have it.

[image]

RA3TOX, 2024-11-18:
This is not the best solution.
It seems we will have to give up this option :(

RA3TOX, 2024-11-18:
@richonguzman
My board is Lora32V2.1 (or V1.6.1)
When I powering the board IO35 read  0 Volt
My firmware from 2024.10.25

richonguzman, 2024-11-18:
it seems your board is bad.. as it should read battery voltage on pin35 with battery connected.

RA3TOX, 2024-11-18:
> your board is bad..

Possibly. 
But @S57PNX also shows incorrect voltage.

richonguzman, 2024-11-18:
> @richonguzman
> 
> > your board is bad..
> 
> Possibly. But @S57PNX also shows incorrect voltage.

this is reading the pin directly (and you have to multiply by two and add a minor correction for the board to get the real value)

the lora32v2.1 board has internal 1/2 voltage divider with two 100K resistors. 

So @S57PNX is having a correct reading of the voltage it enters pin35