richonguzman/LoRa_APRS_iGate #195: Heltec Lora 32 V2 LCD not working

mprilepok, 2024-11-19:
Hi,
I installed version 2024.11.16 on a board Heltec Lora 32 V2 and the LCD is not working although it is configured.

`"display": { "alwaysOn": true, "timeout": 5, "turn180": false },`

[image]

richonguzman, 2024-11-19:
hi @mprilepok ,

does all else work with you?

there are at least 2 versions which are common and one more not that not so common of this HELTEC v2 board. Heltec can have many versions of the same board.

can you double check which version of the Heltec V2 do you have? maybe it needs a pin correction and that's all

mprilepok, 2024-11-19:
This is the board 
Heltec WIFI Lora Kit 32 V2 433MHZ ESP32 LoRa SX1278 esp32 0.96 Inch OLED

PLATFORM: Espressif 32 (6.8.1) > Heltec WiFi LoRa 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash

[image]
[image]

Wifi working, but lora seems to be broken.

mprilepok, 2024-11-19:
this should be the board

https://resource.heltec.cn/download/WiFi_LoRa_32/V2/WiFi_LoRa_32_V2(433%2C470-510).PDF

richonguzman, 2024-11-19:
can you flash it with github repository?

go into /src/display.cpp and change line 24 to look like this:

#if defined(HELTEC_V3) || defined(HELTEC_WS) || defined(HELTEC_V2)

and then just flash the firmware (with the arrow on platformio+vscode) to test?

mprilepok, 2024-11-19:
It did not helped.

I tried other firmware for lora igate, which I used previously and it usus these settings for platformio.

```
board = ttgo-lora32-v1
build_flags = 
  ${esp32_base.build_flags} -DTLORA_V1 -DRESET_OLED=16 -DUSE_SX1278
```

richonguzman, 2024-11-19:
> It did not helped.
> 
> I tried other firmware for lora igate, which I used previously and it usus these settings for platformio.
> 
> ```
> board = ttgo-lora32-v1
> build_flags = 
>   ${esp32_base.build_flags} -DTLORA_V1 -DRESET_OLED=16 -DUSE_SX1278
> ```

those are the same setting im using on this.

this is strange as others have confirmed it worked for them this way too.

you say oled and lora dont work?

can you show the messages it sends over Serial.output when booting after flashing firmware?

mprilepok, 2024-11-19:
Starting Station: OM1PU-12   Version: 2024.11.16
(DigiEcoMode: OFF)
Initializing LoRa Module
init : LoRa Module    ...     done!

Connecting to WiFi 'RadioLAN_32B3' ...
.Connected as 192.168.1.60 / MAC Address: 78:21:84:99:76:10
Connecting to APRS-IS ...     Connected!
(Server: euro.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to RF --
---> LoRa Packet Tx    : OM1PU-12>APLRG1,WIDE1-1:!L6.N9Rk5Ea  GLoRa iGATE & Digi
-- Sending Beacon to APRSIS --
-- Sending Beacon to RF --
---> LoRa Packet Tx    : OM1PU-12>APLRG1,WIDE1-1:!L6.N9Rk5Ea  GLoRa iGATE & Digi
<--- LoRa Packet Rx    : OM1PU-11>APLRT1,WIDE1-1::OM1PU-11 :EQNS.0,0.01,0
(RSSI:-53 / SNR:9.50 / FreqErr:-3091)
---> Uploaded to APRS-IS
<--- LoRa Packet Rx    : OM1PU-11>APLRT1,WIDE1-1::OM1PU-11 :UNIT.VDC
(RSSI:-53 / SNR:11.75 / FreqErr:-3085)
---> Uploaded to APRS-IS
---> LoRa Packet Tx    : OM1PU-11>APLRT1,OM1PU-12*::OM1PU-11 :EQNS.0,0.01,0
---> LoRa Packet Tx    : OM1PU-11>APLRT1,OM1PU-12*::OM1PU-11 :UNIT.VDC
Received Object from APRS-IS  : OM0ODB-S>APDG01,TCPIP*,qAC,OM0ODB-GS:;OM0ODB B *191600z4810.66ND01702.78EaRNG0001/A=000098 70cm Voice (D-Star) 438.32500MHz -7.6000MHz
---> LoRa Packet Tx    : OM1PU-12>APLRG1,WIDE1-1:}OM0ODB-S>APDG01,TCPIP,OM1PU-12*:;OM0ODB B *191600z4810.66ND01702.78EaRNG0001/A=000098 70cm Voice (D-Star) 438.32500MHz -7.6000MHz
<--- LoRa Packet Rx    : OM1PU-11>APLRT1,WIDE1-1:>https://github.com/richonguzman/LoRa_APRS_Tracker 2024.09.18
(RSSI:-55 / SNR:10.00 / FreqErr:-3095)
---> Uploaded to APRS-IS
---> LoRa Packet Tx    : OM1PU-11>APLRT1,OM1PU-12*:>https://github.com/richonguzman/LoRa_APRS_Tracker 2024.09.18
-- Sending Beacon to APRSIS --
-- Sending Beacon to RF --
---> LoRa Packet Tx    : OM1PU-12>APLRG1,WIDE1-1:!L6.N9Rk5Ea  GLoRa iGATE & Digi

mprilepok, 2024-11-19:
now firmware is build with profile ttgo-lora32-v21

mprilepok, 2024-11-19:
I am using this settings, parsswords and ssid is removed
`{
    "callsign": "OM1PU-12",
    "wifi": {
        "autoAP": {
            "password": "1234567890",
            "timeout": 10
        },
        "AP": [
            {
                "ssid": "my-ssid",
                "password": "password"
            }
        ]
    },
    "beacon": {
        "latitude": 48.1631,
        "longitude": 17.0967,
        "comment": "LoRa iGATE & Digi",
        "interval": 15,
        "overlay": "L",
        "symbol": "a",
        "path": "WIDE1-1",
        "sendViaAPRSIS": true,
        "sendViaRF": true
    },
    "aprs_is": {
        "active": true,
        "passcode": "00000",
        "server": "euro.aprs2.net",
        "port": 14580,
        "filter": "m/20",
        "messagesToRF": true,
        "objectsToRF": true
    },
    "digi": {
        "mode": 3,
        "ecoMode": false
    },
    "lora": {
        "txFreq": 433775000,
        "rxFreq": 433775000,
        "spreadingFactor": 12,
        "signalBandwidth": 125000,
        "codingRate4": 5,
        "power": 20,
        "txActive": true,
        "rxActive": true
    },
    "display": {
        "alwaysOn": true,
        "timeout": 5,
        "turn180": false
    },
    "battery": {
        "sendInternalVoltage": false,
        "monitorInternalVoltage": false,
        "internalSleepVoltage": 2.9,
        "sendExternalVoltage": false,
        "externalVoltagePin": 34,
        "monitorExternalVoltage": false,
        "externalSleepVoltage": 10.9,
        "voltageDividerR1": 100,
        "voltageDividerR2": 27,
        "sendVoltageAsTelemetry": true
    },
    "wxsensor": {
        "active": false,
        "heightCorrection": 0,
        "temperatureCorrection": 0
    },
    "syslog": {
        "active": false,
        "server": "192.168.0.100",
        "port": 514
    },
    "tnc": {
        "enableServer": false,
        "enableSerial": false,
        "acceptOwn": false
    },
    "ota": {
        "username": "",
        "password": ""
    },
    "webadmin": {
        "username": "admin",
        "password": "lima1985"
    },
    "ntp": {
        "gmtCorrection": 1
    },
    "other": {
        "rememberStationTime": 30,
        "lowPowerMode": false,
        "lowVoltageCutOff": 0,
        "backupDigiMode": false,
        "rebootMode": false,
        "rebootModeTime": 6
    },
    "personalNote": ""
}`

richonguzman, 2024-11-19:
so it is working with firmware and lora and wifi... only the oled is the problem?

mprilepok, 2024-11-19:
Yes, the oled is not working and i do not see igated trafic.

this id igate https://aprs.fi/info/OM1PU-12
and this is tracer https://aprs.fi/info/OM1PU-11

both devices are in rage.

richonguzman, 2024-11-19:
OM1PU-11 is tracker
OM1PU-12 is igate

what station is your heltec-v2 ?

mprilepok, 2024-11-19:
OM1PU-12 is igate is running on heltec-v2

richonguzman, 2024-11-19:
> OM1PU-12 is igate is running on heltec-v2

pass_code = 22783 ?

all the messages you got from serial output are only available because it got connected to aprs-is. servers

so now this is more strange

mprilepok, 2024-11-19:
yes, in the passcode there was an invisible char, now wifi and lora is working. Trafic is gated  from rf to inet and visevera.

the one thing is broken, OLED is not working.

richonguzman, 2024-11-19:
> yes, in the passcode there was an invisible char, now wifi and lora is working. Trafic is gated from rf to inet and visevera.
> 
> the one thing is broken, OLED is not working.

good news, so now we know that only this is missing... lets continue:

1) did you test the board with other firmwares and does it work ?
2) if (1) is yes, please tell me which link and firmware to compare exactly what to change or see to compare them

richonguzman, 2024-11-19:
> https://github.com/lora-aprs/LoRa_APRS_iGate

I know this firmware, but tlora-v1 is another board than yours... so now I'm more confused

mprilepok, 2024-11-20:
i looked at the board under microscope, it looks like Heltec WiFi LoRa 32 V2 board https://docs.heltec.org/en/node/esp32/wifi_lora_32/hardware_update_log.html

richonguzman, 2024-11-20:
i validated all the pins and all I could: they are in the right position.

I've added a minor update to the main code: can you upload the latest from github (not from webflasher) just the firmware?

richonguzman, 2024-11-20:
> oled is still not working

added one more thing to main repository: I thing I got it this time:

adding "|| defined(HELTEC_V2)" to /src/power_utils.cpp in line 208

please download the lates main from github and upload just firmware to validate

mprilepok, 2024-11-20:
You are king, It works.
[image]

richonguzman, 2024-11-20:
Great news!!!! Finally

closed 2025-01-09 by richonguzman