richonguzman/LoRa_APRS_iGate #199: Question: How do I setup 2024.11.06 for digi only?

AG7U, 2024-11-30:
I am currently using version 2024.01.12 for digi.  On the screen I see 
     Rx: 433.775 Tx: 433.775
     <<   DigiRepeater   >>
The above on-screen message lets me know the TTGO-LoRa32 is operating as a digi.

When I attempt to setup 2024.11.06 for digi only I cannot replicate the above screen, and I am not sure if digi is setup correctly.  

Is the igate_conf.json below correct for digi only?
In version 2024.01.12 should I expect to see a screen like I see in version 2024.01.12?  I do eventually see LoRa RX ----> LoRa Tx, which would suggest that digi is working.  
How do I edit igate_conf.json to get the digipeater simplex green star?
        "overlay": "L",
        "symbol": "a",
Can the on-screen WiFi: AP APRS-IS: OFF IP: 192.168.4.1 message be removed from the screen by toggling a setting in the igate_conf.json?
Anything else I should do or know for digi only using version 2024.11.06?  
Thank you.  

{
    "callsign": "AG7U-40",
    "wifi": {
        "autoAP": {
            "password": "1234567890",
            "timeout": 10
        },
        "AP": []
    },
    "beacon": {
        "latitude": 39.529639,
        "longitude": -104.824333,
        "comment": "LoRa APRS",
        "interval": 15,
        "overlay": "L",
        "symbol": "a",
        "path": "WIDE1-1",
        "sendViaAPRSIS": false,
        "sendViaRF": true
    },
    "aprs_is": {
        "active": false,
        "passcode": "XYZVW",
        "server": "rotate.aprs2.net",
        "port": 14580,
        "filter": "m/10",
        "messagesToRF": false,
        "objectsToRF": false
    },
    "digi": {
        "mode": 2,
        "ecoMode": false
    },
    "lora": {
        "txFreq": 433775000,
        "rxFreq": 433775000,
        "spreadingFactor": 12,
        "signalBandwidth": 125000,
        "codingRate4": 5,
        "power": 20,
        "txActive": true,
        "rxActive": true
    },
    "display": {
        "alwaysOn": true,
        "timeout": 4,
        "turn180": false
    },
    "battery": {
        "sendInternalVoltage": false,
        "monitorInternalVoltage": false,
        "internalSleepVoltage": 2.9,
        "sendExternalVoltage": false,
        "externalVoltagePin": 34,
        "monitorExternalVoltage": false,
        "externalSleepVoltage": 10.9,
        "voltageDividerR1": 100.0,
        "voltageDividerR2": 27.0,
        "sendVoltageAsTelemetry": false
    },
    "wxsensor": {
        "active": false,
        "heightCorrection": 0,
        "temperatureCorrection": 0.0
    },
    "syslog": {
        "active": false,
        "server": "192.168.0.100",
        "port": 514
    },
    "tnc": {
        "enableServer": false,
        "enableSerial": false,
        "acceptOwn": false
    },
    "ota": {
        "username": "",
        "password": ""
    },
    "webadmin": {
        "username": "admin",
        "password": ""
    },
    "ntp": {
        "gmtCorrection": 0
    },
    "other": {
        "rememberStationTime": 30,
        "lowPowerMode": false,
        "lowVoltageCutOff": 0,
        "backupDigiMode": true,
        "rebootMode": true,
        "rebootModeTime": 12
    },
    "personalNote": ""
}

richonguzman, 2024-12-01:
Can you edit all through the WebConfig UI created for all of this?

uploading from VSCODE filesystem igate_conf.json may lead to missing a variable or editing errors.

symbol is selected over WebConfig UI without any problem.
but if you need green star and still do it manually -->  "symbol": "#",

at last, display should show DIGIREPEATEr without any problem, let me know if this is shown on yours

AG7U, 2024-12-01:
Using VSCODE I erased flash and performed a full clean.  Using WebConfig UI I configured TTGO-LoRa32 for digipeater WIDE1.  I do not see the word DIGIREPEATER on the LoRa32 screen.  Attached is the WebConfig UI configuration.  What am I doing wrong?  This digipeater will be going back to an environment with no WiFi, which is why I did not update WiFi Access or APRS-IS settings.

richonguzman, 2024-12-01:
but can you confirm its repeating? (besides the screen problem?)

closed 2024-12-31 by richonguzman