richonguzman/LoRa_APRS_iGate #200: igate not reconnecting to Wi-Fi after wake from sleep

pdevlin13, 2024-12-06:
My remote iGate running main branch build 20th November, does not reconnect to WiFi after waking from sleep but wakes into Digipeater backup mode.
I had the same problem a couple of weeks ago, I thought the problem may have been that my 4G router had timed out with no connection. However when I visited the site I noticed my automaticly connected to the 4G router so no issue there.
I updated to the latest build in the main branch as of 20th November hoping this may fix the issue but the problem still exists.
I have the igate set to reboot every 24 hours, but it reboots different some how after waking, it does not transmit the the Beacon comment with GitHub link that you normally get on a restart. It as though it inst actually restarting.
Also the 30 minute beacon becomes 31 or 32 minutes, see attached imag.
Thanks
Paul
MW0PDV
[image]

richonguzman, 2024-12-09:
hi, I got confused:

there a lot of things going on here.

how far from Wifi is the igate board?

is it now rebooting at the exact time ? (how big is the error in seconds)?

sending each 30 min , how big is the tx offset?

pdevlin13, 2024-12-09:
Hi Ricardo, 
The igate (Heltec v3) is about 2m from the 4G router, this a remote iGate. 
I should have mentioned that I have Backup Digipeater enabled.

Due to the very short daylight hours in the UK currently, the board enters sleep not long after sunset and so it isn’t currently reaching 24hours for a timed reboot. The wake time is obviously variable depending on sunlight for solar.
The beacon time is set to 30 minutes but the beacon in Digipeater backup is actually  31 minutes and  1 second or 1,861 seconds.
When it was doing this a couple of weeks ago I visited the remote site and a power cycle of the igate seemed to fix it, until the next time it entered sleep.

hope that helps
Paul.

richonguzman, 2024-12-09:
so the backup digimode sends all packets with 31:01 min:sec delay from last packet or just the first one?

pdevlin13, 2024-12-09:
so the backup digimode sends all packets with 31:01 min:sec delay from last packet or just the first one?

Once in backup mode every beacon is 31:01

richonguzman, 2024-12-09:
so this is working at as a Digi, but when it receives much packets to repeat it adds the self beacon at the list of packets to Tx.

so maybe this delay is a good amount of packets being repeated before the Station GPS beacon Tx ?

pdevlin13, 2024-12-09:
It is a very remote igate, the only traffic it normally will see is someone on a nearby SOTA summit. I added the 4G router as it cannot reliably reach my home, or another local igate. 
 Should the igate reboot when waking from sleep? It’s as though when it wakes from sleep it doesn’t  exit backup mode.

richonguzman, 2024-12-09:
when it exit SleepMode is only when voltage is high enough. So maybe the delay added is because it needs to re-connect to the 4G Router?

im trying to see/think how to test a similar process here

pdevlin13, 2024-12-09:
Hi Ricardo, just be to clear the main problem is that it will not reconnect to Wi-Fi on waking, an so remains in backup mode.
I mention the beacon time in the hope this will help to diagnose the problem.

richonguzman, 2024-12-09:
now I get the scenario more:

as it cant connect to wifi it will get into backup mode. (the delay is the time it tries to connect to the 4G router and fails)

so the problem is WHY IT FAILS? right?

does it reconnect to Wifi at anytime after entering into digiBackUpMode?

pdevlin13, 2024-12-09:
Hi Ricardo 
No once it’s entered backup mode it does not connect to Wi-Fi 

I initially assumed the 4G router had shutdown the first time this happened, but when I visited the remote site my phone immediately connected to the 4G router with good internet.

richonguzman, 2024-12-09:
so we should check about the Wifi connection there. how many Wifi ssid are on the igate/digi station registered?

is the router for 2.4GHz wifi only?

richonguzman, 2024-12-09:
will check for info about this problem as it should work without any issues

pdevlin13, 2024-12-09:
Thanks Ricardo 
Just to maybe help diagnose the problem. When the igate wakes, there is no beacon txt with your Githhub link and build date, like you see on a restart.

pdevlin13, 2024-12-10:
Hi Ricardo, I have emulated the remote igate setup except using my home LAN and everything works fine.
It is pointing to something to do with waking up the 4G router. It fails to connect to the router  on  wake from sleep, but will connect on a power cycle of the board. Maybe I can add a delay or increase the Wi-Fi reconnect attempts?

Thanks
Paul

richonguzman, 2024-12-10:
Once I added a delay to start again after boot but this must be flashed from vscode to test

pdevlin13, 2024-12-10:
Hi Ricardo, thanks, I am using OTA updates so will create a new bin file in VS code and give it a try.
It will be few days before I can get to the site to try it.

Thanks
Paul.

pdevlin13, 2024-12-11:
Hi Ricardo I will close this issue. The issue seems to be the router. Looking at the code I can see the 1:01 delay is 2 attempts at of 30 seconds trying to reconnect Wi-Fi before dropping back to Digi backup. I also realised that it worked when I cycled the power as my phone had already woke the router. Still not sure why my phone wakes the router but esp32 doesn’t. I will try a different router. 
Thanks
Paul.

closed 2024-12-11 by pdevlin13

richonguzman, 2024-12-11:
> Hi Ricardo I will close this issue. The issue seems to be the router. Looking at the code I can see the 1:01 delay is 2 attempts at of 30 seconds trying to reconnect Wi-Fi before dropping back to [...]

would love to see a way to validate that esp32 wakes up router too 

lets hope we find a way.