richonguzman/LoRa_APRS_iGate #203: Heltec Esp32 V3 TX non stop

nazainurin, 2024-12-10:
Im using latest firmware for Heltec Esp32 V3 and it TX non stop. On lcd screen appeared TX inforfamtion and not back to normal. What can I do?

[image]

richonguzman, 2024-12-10:
This is tracker firmware?

nazainurin, 2024-12-10:
> This is tracker firmware?

yes. latest firmware for tracker. If im use for iGate, this board always restart when received the signal. But now I want to use for tracker. My board already install GPS module.

richonguzman, 2024-12-30:
Hi, could you test it further?

richonguzman, 2025-01-09:
this should been solved with the latest updates for Heltec boards

nazainurin, 2025-01-09:
still same

richonguzman, 2025-01-09:
which version of V3 is your board? V3 or V3.2 ?

you have this board flashed with Tracker or iGate firmware?

nazainurin, 2025-01-09:
im working as tracker
*MOHD NAZAINURIN BIN MOHD NAWAWI*
*9M2OCW / 9M4BCW / 9M4GCW*
*Malaysia Amateur Radio Operator*

*No 18, Jalan Pulasan 32,*
*Taman Kota Masai,*
*81700 Pasir Gudang, *
*Johor Darul Ta'zim.*

*Email : ***@***.*** / ***@***.****
*No tel : 011-1072 2904 / 011-1214 1521*

richonguzman, 2025-01-10:
hi, lets work it as a tracker:

which firmware version did you use?

HELTEC_V3_GPS
HELTEC_V3_TNC??

is yours new HELTEC_V3.2?

closed 2026-09-30 by richonguzman