richonguzman/LoRa_APRS_iGate #206: Eco Mode exit for HT-CT62 not working

MihaiRF, 2024-12-20:
Hello,
I have just tested the latest release with my HT-CT62 module. Entering the "Eco Mode" using APRS queries works (the board consumes about 14mA@3V). However, when I try to wake it up so that I can have the wi-fi AP back again, I cannot connect to the board. The consumption goes to ~60mA (insted of ~100mA with the AP on). The wi-fi seems dead and the board unresponsive. Restarting the power gets me back into Eco Mode.

PS. Refering to the consumption in Eco Mode, just wanted to highlight that it is considerably higher than the _Low power mode (only for HT-CT62)_ tested some time ago, that was around 8 mA.

richonguzman, 2024-12-20:
LowPowerMode works awesome and soon all boards will be updated to this kind of work (sleeping the esp32 but keep the LoRa module waking the esp32 to process the packet)

Did not try the EcoMode on this board , but it should wake up the wifi without any issues as all the other boards.

maybe its that it get not saved on internal memory the change?

was the Wifi info available in internalmemory before?

please add all the configuration data you can to understand where to look for posible fails

MihaiRF, 2024-12-20:
So, I had some random brownout resets and I was able to find the source of this problem, but unfortunately this did not solved the issue. The LDO I'm using to get the 3.0V is not providing a steady 3.0V output so I just powered directly with 3.0V. It solved the random resets but the issue remains. Here is the output starting from sending the "?APRSSEM" command:

```
<--- LoRa Packet Rx    : YO5OMT-12>APLRG1,WIDE1-1:}YO5OMT-5>APDR16,TCPIP,YO5OMT-12*::YO5OMT-15:?APRSSEM{8
(RSSI:-88 / SNR:6.50 / FreqErr:3797)
---> LoRa Packet Tx    : YO5OMT-15>APLRG1,WIDE1-1::YO5OMT-5 :ack8
---> LoRa Packet Tx    : YO5OMT-15>APLRG1,WIDE1-1::YO5OMT-5 :DigiEcoMode:Start
Saving config..
Config saved
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381e64
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6a0
load:0x403ce710,len:0x2624
entry 0x403cc710
[   335][E][esp32-hal-cpu.c:207] setCpuFrequencyMhz(): CPU clock could not be set to 240 MHz

Starting Station: YO5OMT-15   Version: 2024.12.06
(DigiEcoMode: ON)
Initializing LoRa Module
init : LoRa Module    ...     done!
<--- LoRa Packet Rx    : YO5OMT-12>APLRG1,WIDE1-1:}YO5OMT-5>APDR16,TCPIP,YO5OMT-12*::YO5OMT-15:?APRSEEM{4
(RSSI:-87 / SNR:6.50 / FreqErr:3796)
38304Reconnecting to WiFi...

Not connected to WiFi! Starting Auto AP

Not connected to WiFi! Starting Auto AP
```
The board is not responding to any other queries after that and consumes about 60mA. Wi-fi is down as well.

MihaiRF, 2024-12-20:
I just wanted to highlight again how useful it would be (at least in my case) to have the experimental option "_Low power mode (only for HT-CT62)_" available to switch OFF/ON without the serial interface. 
Just checked again and the consumption is 8.1mA when enabled. Las month I have tested how long it lasts without sun (solar panel) and it went for almost 3 weeks with TX beacon at 15minutes and little traffic (2 x 18650 Li-ion cells in parallel).

SQ2CPA, 2024-12-20:
@MihaiRF I'm a creator of low power mode for HT-CT62. This mode works only for Digi, WiFi is not possible and it have separate part in the code with separate logic. Any other functions will not work in that mode because that will require more power. As you said, power consumption is very good and I don't want to destroy that by implementing anything extra that will use more CPU time etc. I have some testing gates (SQ2CPA-14, SQ2CPA-1, also SR2LMU-10) that works fine for month with 100% battery percentage. You can exit low power mode by sending message `?APRSELP` to your gate, this will exit low power mode and run WiFi etc. so you can configure or update your gate.

MihaiRF, 2024-12-20:
@SQ2CPA
You've made my day :)
Just tested that, and it works waking up the Wifi so I can edit the configuration without the wires :)
I needed this mostly because I want to be able to update the location coordinates after installing into final location. The location is uncertain for now and it's nice to be able to change it rapidly once found (best mountain top / highest tree :).
Thanks again for the help :)

SQ2CPA, 2024-12-20:
@MihaiRF That's must have for that mode. We have gates where we can't access anything and there is no way to open the case to update the software or change configuration. I will also make PR update for that mode in few days with telemetry. You can check that on SQ2CPA-10. This board is running in my room without external power, I just want to check again what will be the battery voltage chart.

MihaiRF, 2024-12-20:
@SQ2CPA
That's good news. 
I'm measuring the battery voltage through 27K & 27K resistors divider, connected to pin no. 1 on the module. Leakage current is acceptable. In order to get the exact voltage reading, I have chosen these two values in the web config: R1=120K; R2=161K. The reported voltage is pretty accurate from 3.3V to 4.2V and I'm satisfied with that.

S57PNX, 2024-12-21:
@SQ2CPA can you run BME280 WX on HT-CT62 Low power mode or is this one of those things that will increase the consumption significantly?

SQ2CPA, 2024-12-21:
@S57PNX Not yet but sooner or later I will do that because I know that this will be interesting and many people asked about that. BME280 have also sleep mode but that's not a problem for me. Only problem is to mount this sensor outside the box leaving the structure intact

S57PNX, 2024-12-21:
@SQ2CPA  for indoor deployments I just run 4 wires outside of the box and put the BME280 in a small enclosure from Thingiverse: https://www.thingiverse.com/thing:4778662  For outdoor, I usually just drill a 4mm hole in my waterproof box and install the BME280 board flat from the inside so that only the tiny sensor chip is in the hole. Then a generous amount of hot glue holds everything in place and hopefully sealed somehow. I apply a coat of clear lacquer on the BME280 board beforehand to make it more moisture-proof.

SQ2CPA, 2024-12-22:
@S57PNX I just bought modified PCB with I2C exposed so I will play with that after 2-3 weeks, I will implement WX for low power mode. 

@richonguzman I think that this issue is resolved and you can close

closed 2024-12-30 by richonguzman